Bodmer/TFT_eSPI #89: Can't draw character with x or y coordinates < 0

tonywestonuk, 2018-02-03:
My application needs to show large numbers.... I have converted a 128 point font  into progmem bitmaps.

However, when I display the font at co-ordinates 0,0 the bottom of the character is below the bottom of the screen, while there is a large gap above the font.   If I could draw the font at say position -20 y it would position correctly in the centre of the screen.  (see attached jpg)

Thanks.

[image]

Bodmer, 2018-02-03:
Which function are you using to render the bitmap?

tonywestonuk, 2018-02-04:
drawCentreString

tonywestonuk, 2018-02-04:
Attached is my arduino project.  To be fair this issue is also in the adafruit drivers. 
[BarberQ.zip](https://github.com/Bodmer/TFT_eSPI/files/1692584/BarberQ.zip)

Bodmer, 2018-02-04:
The font file format is not correct. How did you generate it?

The parameters are:	
bitmapOffset, width, height, xAdvance, xOffset, yOffset

The yOffset should be a negative number, for a digit the magnitude is typically about the same as the digit height. I would guess you need to subtract 256 from the number you have, e.g. instead of:
  {     0, 103, 188, 118,   12,   69 },   // 0x30 '0'
it probably should be:
  {     0, 103, 188, 118,   12,    -187},   // 0x30 '0'

Bodmer, 2018-02-05:
I note the yOffset is and 8 bit signed value so cannot hold -187!

I have had a play and fixed the font file and adjusted the values so the characters do not jump about so much. Some segments are in different positions (e.g. 5 compared to 8) so some movement of segments will happen.

See and try
[bQ.zip](https://github.com/Bodmer/TFT_eSPI/files/1693453/bQ.zip)
 attached.

tonywestonuk, 2018-02-05:
Thankyou! -  very appreciated. :-)

The original font I made by converting a true type font (I think it was this one http://torinak.com/7segment) to bitmaps with adafruit fontconvert.

I'll try this out later on when I get back home.  Thanks again.
:-)

Bodmer, 2018-02-05:
The problem is with the limitation of the Adafruit font format due to the use of the 8 bit signed offset. As this is not a library bug I will close this issue.

Good luck with your project.

closed 2018-02-05 by Bodmer