Bodmer/TFT_eSPI #90: ILI9341 in overlap mode (sharing SPI0 port)

JamesNewton, 2018-02-08:
So this works great on the ST7735 display, (see Issue #74) but I can't see to get it going for the ILI9341. Figure I should check:

Is there any reason the ILI9341 won't work in overlap mode?

First, hardware is NodeMCU and Adafruit
https://www.adafruit.com/product/2478
All works perfectly with `Setup1_ILI9341.h`file selected in `User_Setup_Select.h` and the standard wiring. I'm doing this because I want a scrolling terminal for debug messages for my project as well as a nice big bright DRO numbers display, so I'm starting with your lovely `TFT_Terminal.ino` demo. No problems there. 

I've copied the existing `Setup1_ILI9341.h` file to make a new `Setup12_ILI9341_Overlap.h`  file which I've added to the `User_Setup_Select.h` file and selected. Here it is as a text file
[Setup12_ILI9341_Overlap.h.txt](https://github.com/Bodmer/TFT_eSPI/files/1705293/Setup12_ILI9341_Overlap.h.txt)

````
// ModeMCU
// Do not define TFT_CS in overlap mode, TFT chip select must connect to pin D3
#define TFT_CS   PIN_D3  // GPIO 0
#define TFT_DC   PIN_D5  // GPIO 14 Data Command control pin
#define TFT_RST  PIN_D4  // GPIO 2 Reset pin (could connect to NodeMCU RST, see next line)
//#define TFT_RST  -1  // Set TFT_RST to -1 if the display RESET is connected to NodeMCU RST or 3.3V
````

I've tried with the PIN_D3 line in or out, and it does't seem to make a difference. I've triple checked the wiring:
// Display SDO/MISO  to NodeMCU pin SD0 (just like on my ST unit. 
// Display LED       to NodeMCU pin VIN 
// Display SCK       to NodeMCU pin CLK
// Display SDI/MOSI  to NodeMCU pin SD1
// Display DC   to pin D5 //NOT D3
// Display RESET     to RST
// Display CS        to NodeMCU pin D3
// Display GND       to NodeMCU pin GND (0V)
// Display VCC       to NodeMCU 5V or 3.3V

I've also tried setting the SPI_FREQUENCY down to the same speed it was on the ST:
 #define SPI_FREQUENCY  27000000 // Actually sets it to 26.67MHz = 80/3

JamesNewton, 2018-02-08:
Nevermind. Found that there was a missing define:
`#define TFT_SPI_OVERLAP`
Which is in the ST7735 Overlap file which I did not see and so didn't copy over into the new Setup12_ILI9341_Overlap.h file. Seems to me like that define should be up in the section that talks about the pinout for the overlap mode.

Also, the specific Adafruit display
https://www.adafruit.com/product/2478
has some mode pins that need to be shorted in order to switch from parallel to SPI operation. 
https://learn.adafruit.com/adafruit-2-4-color-tft-touchscreen-breakout/spi-wiring-test

Looks nice now!

closed 2018-02-08 by JamesNewton

Bodmer, 2018-02-08:
Thanks for pointing this out I will correct in the next update cycle.

JamesNewton, 2018-02-08:
Updates / pull requests ready for you.

Bodmer, 2018-02-09:
This has been addressed in the latest update.