Bodmer/TFT_eSPI #94: TFT_eSPI.h and MFRC522.h slows down TFT

ismaelit, 2018-02-08:
Hello guys. This is my first post here and I hope to contribute in some way.

I'm developing an application using ILI9341 with this spectacular Bodmer TFT_eSPI library. Now I am integrating with an RFID-RC522, including the MFRC522.h library and instantiate without any problems. 

But after starting the RFID sensor - mfrc522.PCD_Init() in the setup() section, the TFT speed drops to the original SPI standards: without acceleration.

Does anyone see any solution to this problem?

Bodmer, 2018-02-08:
You need to enable transaction support in the setup file, see the end of that file.

ismaelit, 2018-02-08:
Crying of happiness! It Works! 
Thank you very much @Bodmer :))

closed 2018-02-08 by ismaelit