Bodmer/TFT_eSPI #95: How to draw bitmaps from SD Card

gamecompiler, 2018-02-10:
How do i draw bitmaps from SD Card?
I didnt got it work. I even didnt got the JPEGDecoder example to work (Changed it to SD card filesystem instead of SPIFFS).

Im using an ESP32.

Bodmer, 2018-02-11:
Does the sd test example in th SD library run OK.?

gamecompiler, 2018-02-11:
Yes, I can mount the SD card correctly and read/write files.

Bodmer, 2018-02-11:
Which pins are you using to connect to the display and SD card?

Can you post a basic sketch that you are trying to get working?

gamecompiler, 2018-02-15:
Yes, really really bad coding style, i can do that better. Displaying a bitmap from sd card was just a small thought in the break of my lectures. Id never coded C++. Only Java.
 This code is completly copied fro the Adafruit ILI9341 library. There is a lot i misunderstood i guarantee.

TFT_eSPI.cpp
```
void TFT_eSPI::startWrite(void){
  spi_begin();
}

void TFT_eSPI::endWrite(void){
  spi_end();
}

void TFT_eSPI::writePixel(uint16_t color){
    spiwrite(color);
}

```

spitftbitmap.cpp
```
#include <Arduino.h>

#define FS_NO_GLOBALS
#include <FS.h>
#include <SD.h>
#include <SPI.h>
#include <TFT_eSPI.h>
#include <Adafruit_GFX.h>
TFT_eSPI tft = TFT_eSPI();

//Code copied from https://github.com/adafruit/Adafruit_ILI9341/blob/master/examples/spitftbitmap/spitftbitmap.ino
//Changed in mind, it MAY display bmp images from the sd card.
void setup(){

  Serial.begin(115200);

  tft.init();
  tft.setRotation(3);

  tft.fillScreen(TFT_BLACK);

  if (!SD.begin()) {
    Serial.println("SD Card initialisation failed!");
    while (1) yield(); // Stay here twiddling thumbs waiting
  }
  Serial.println("\r\nInitialisation done.");

  uint8_t cardType = SD.cardType();

  if(cardType == CARD_NONE){
      Serial.println("No SD card attached");
      return;
  }

  Serial.print("SD Card Type: ");
  if(cardType == CARD_MMC){
      Serial.println("MMC");
  } else if(cardType == CARD_SD){
      Serial.println("SDSC");
  } else if(cardType == CARD_SDHC){
      Serial.println("SDHC");
  } else {
      Serial.println("UNKNOWN");
  }

  uint64_t cardSize = SD.cardSize() / (1024 * 1024);
  Serial.printf("SD Card Size: %lluMB\n", cardSize);

}
[... 146 lines cut ...]
    }
  }

  bmpFile.close();
  if(!goodBmp) Serial.println(F("BMP format not recognized."));
}

void loop(){
  for(uint8_t r=0; r<4; r++) {
    tft.setRotation(r);
    tft.fillScreen(TFT_BLACK);
    for(int8_t i=-2; i<1; i++) {
      bmpDraw("/purple.bmp",
        (tft.width()  / 2) + (i * 120),
        (tft.height() / 2) + (i * 160));
    }
  }
}

```

This is the image i tryed to display:
https://cdn-learn.adafruit.com/assets/assets/000/013/448/original/adafruit_products_purple.bmp?1388768741

The code just renders some random noise on the screen.

gamecompiler, 2018-02-15:
Yup, found two errors just a after posting. Changing the method writePixel:

```
void TFT_eSPI::writePixel(uint16_t color){
    writedata(color);
}
```
I can at least identify the flower.... The next problem is the conversation from 16 to 8 bit i forgot i guess.

gamecompiler, 2018-02-15:
Got it working, but i really dont know why it works now.
```
void TFT_eSPI::startWrite(void){
  digitalWrite(TFT_CS, LOW);
  spi_begin();
}

void TFT_eSPI::endWrite(void){
  digitalWrite(TFT_CS, HIGH);
  spi_end();
}

void TFT_eSPI::writePixel(uint16_t color){
    pushColor(color);
}
```

Please leave this issue open. I thing theres data transmitted where it shouldnt.

gamecompiler, 2018-02-15:
Works....
Just changed:
```
//b = sdbuffer[buffidx++];
//g = sdbuffer[buffidx++];
//r = sdbuffer[buffidx++];
//to
r = sdbuffer[buffidx++];
g = sdbuffer[buffidx++];
b = sdbuffer[buffidx++];
```

Bodmer, 2018-02-22:
The file seek is very slow so the attached approach will probably draw the image ~4x faster:

[TFT_SD_BMP.zip](https://github.com/Bodmer/TFT_eSPI/files/1746283/TFT_SD_BMP.zip)

Drawing jpeg will probably be faster too as the files are smaller.

closed 2018-02-22 by Bodmer

PumaFPV, 2020-01-29:
Hello, I am using an esp32 with 240x240 st7789 but it is not working, i have a white screen with weird black moving lines, any idea why? it is the stock code, i have tried with several pictures, bmp, jpg and png, all of them the size of my screen 
thank you

Bodmer, 2020-01-30:
Since this topic is bitmaps from SD cards, do I assume the other library examples run fine and you are just having trouble with bitmap images?

PumaFPV, 2020-01-30:
Yes I can display anything but image from sd card (I can display images that are stored in the progmem but after 7 images, my memory is full) thus the need for using an sd card

Bodmer, 2020-01-30:
Have you run the SD library sketches successfully with your SD card setup?

If yes:
Which version of TFT_eSPI are you running?
Which version of the Arduino IDE?
Which version of the ESP32 board package.
Which SD library are you using?
Which example sketch are you trying to run?
Is anything reported to the Serial Monitor window by your test sketch, if so what?

PumaFPV, 2020-02-02:
Yes my SD card is working properly alone.
TFT_eSPI: 1.4.20
Arduino IDE: 1.8.10
ESP32 package: ESP32-WROOM-32 seems to be version 1.0.4
SD lib: included with esp32 package
Serial output: Loaded in 526 ms
That's it ^^'

Bodmer, 2020-02-02:
Which example sketch are you trying to run?

PumaFPV, 2020-02-02:
TFT_SD_BMP
I've tried with bmp, jpg and png images

Bodmer, 2020-02-02:
What are your connections from the ESP32 to the TFT and also to the SD Card?

PumaFPV, 2020-02-02:
SPI, each work independently, but it seems  like the drawBmp function doesnt work properly here

Bodmer, 2020-02-02:
Try a lower SPI clock rate in your setup file, say 10MHz and test that.

PumaFPV, 2020-02-02:
Now I get a black screen, i dont know if it means any good

Bodmer, 2020-02-03:
Which pins have you used for the chip select on the SD card and display?

PumaFPV, 2020-02-03:
sd card cs: 4
st7789 cs: 5

PumaFPV, 2020-02-04:
Found an issue that might be linked, whatever #define TFT_DS I use, only the pin 5 will work, and nothing else

Bodmer, 2020-02-05:
Thi si te way the SD library works. If you use:
```
 SD.begin();
```
then the default is pin 5.
If you use this:
```
SD.begin(SD_CS); // or say SD.begin(22) for gpio 22
```
Then the SD_CS pin you have defined will be used as the chip select.

PumaFPV, 2020-02-05:
Yes I know that, I've choosen the pin 4 so SD.begin(4);
And this works fine since the SD card is working with other sketches, and the tft too... I would think that the problem comes from the conversion of the image from the sd to the tft, as if either my images are not compatible or if doesnt convert them the right way

Bodmer, 2020-02-06:
In that case I do not understand what you mean by "whatever #define TFT_DS I use, only the pin 5 will work". The example TFT_SD_BMP uses:
```
#define SD_CS 22
```
at line 16...

The example TFT_SD_BMP came with an image parrot.bmp, have you tried that?

Bodmer, 2020-02-06:
Note that in some pre-defined user setups pin 4 is used as the TFT_RST (reset)! List all ESP32 pins you have used and what they are connected to.

Make sure the power supply is adequate to power everything. Make sure you have good electrical connections for GND to all boards, preferably GND lines should be short and not use thin wire.

Bodmer, 2020-02-06:
I just ran the example from the zip file attached to the post on 22/2/18 and it ran fine with Setup24_ST7789.h

mcufriend, 2020-09-17:
Thanks, Bodmer ，
 reading  BMP     images file from SD card, the display speed is greatly improved. I used  TFT-ESPI library, 3.5 "SPI display 480 X 320  ，Especially the one that was provided at the end，“TFT_SD_BMP.zip”  ，And it's very effective,，Originally wanted to abandon BMP and use JPEG files directly.

mcufriend, 2020-09-17:
stm32f103rbt6  + 3.5 inch tftlcd  (spi ),  cpld -spi   ,480  x 320 
tft-espi,   guislice ,

Bodmer, 2021-05-09:
Yes, same pins, use 18,19,23 for SPI bus. Get SD library examples working first with the TFT_CS signal pulled or driven to logic 1. Then try that example.

Note that removing power while an SD card read is in progress can corrupt the SD card files, so then the card needs to be reformatted and files written again.

Anxodia, 2021-08-29:
Hallo Bodmer.

I am really sorry to bring this up again. I have been browsing the web the past three days not finding a solution to my problem. I've seen the many things that you have already achieved so maybe it is easy to provide an answer for me.

I am currently studying mechanical engineering and Im woking on my diploma-thesis at the moment. My project involves a 3.5inch 320x480 touch screen HX8357D from Adafruit with SD Card (in SPI mode) and an Arduino Mega. 

I need to draw bitmaps and have a menu with buttons that you can click trough. The menu needs to look pretty clean so I designed the menu-screens bitmaps myself. Just drawing texts isnt enough for my project.

My problem is, as to be expected loading time of the BMP24. Its just so clunky to navigate trough the menu at that speed.

I read about raw files, PNG conversion and all that stuff but I didnt find anything that suits my HX8357D.

Is there a method to achieve faster draw speed with reasonable amount of effort? I have no experience in coding btw, this is my first time. Im building a micrometer precise XY-Stage which goes under a microscope and its controlled trough the touch panel menu.

My menu screens only have three colours so maybe drawing a BMP16 oe even 8 would do the job, if thats possible?

Im happily awaiting your response
Thanks im advance,
Louis