Bodmer/TFT_eSPI #96: screen content is mirrored (ILI9341)

maxSnyder, 2018-02-20:
Hi bodmer,
first of all let me say, that you've done an excellent job with your TFT_eSPI library for the esp8266. I'm using it with a Wemos D1 R2, which is equipped with the esp8266, too, but this board is in the Arduino UNO layout, where the pin headers have another pin assignment than the NodeMCU ESP-12, whose pin assignment is defined in the User_Setup_Select.h.
[image] Wemos D1 R2

After an adaptation to these changed pin conditions, your library worked (almost) great at the first attempt, the TFT as well as the touch controller.
I said "(almost)", because there is a not insignificant restriction:
In all but one rotation settings 0 … 7, the screen content is mirrored, either symmetrically to the central point or to one oft the middle axes. And therefore the x, y - coordinates of the touch controller appear at a different (mirrored) place on the screen and not there, where the screen was touched. So for example the touching of a screen button cannot be recognized.
I'm using this 2.4" TFT display with the ILI9341 and the XPT2046 touch screen controller:
 
[image]

Maybe you have an idea, what's going wrong here?
thanks in advance

Bodmer, 2018-02-20:
Different suppliers wire up the TFT pixels to the controller in different ways, hence the problem. All my displays work ok.

There is an option to correct this by using:
#define M5STACK
In the setup, but in your case it will not compile unless an arbitrary free pin is defined for the backlight, say:

#define TFT_BL   0

Try it if you have time, I will look for a better way and update the library to cope with these reversed displays.

maxSnyder, 2018-02-21:
I tried your suggested option, but it does not work: the screen content is still mirrored, but now the screen is smaller, in rotation 1 (portrait) is a white field in the lowest quarter of the screen and when running your touch calibration sketch, the markings on the left side are out of screen.

Bodmer, 2018-02-21:
OK, later today I will create a sketch for you to run so we see hat is happening and can work out the required settings for the TFT control register in your display.

Bodmer, 2018-02-21:
I've created the attached sketch so you can find the settings I need to define the rotation.

Read the comments at the start on how to use it.  I need 4 control register values, one for each rotation.

[MAD_CTL.zip](https://github.com/Bodmer/TFT_eSPI/files/1745908/MAD_CTL.zip)

maxSnyder, 2018-02-22:
Your sketch provided the following result:
From the 8 combinations which are possible with the 3 bits y, x, v, for every rotation 0 - 3, I got 4 readable text strings, 2 in portrait and 2 in landscape format:

[image]

But **directly after a power on or a reset**, the screen shows this content for e.g. MAD_CTL = 0x48 in rotation 0:
[image]
Only after changing x two times, the result was the one shown in the very first picture.

Bodmer, 2018-02-22:
According to the results you have a standard screen setup... and the library should work fine as is. Something strange is happening in your setup, so I think we have to look elsewhere than the software.  Do you have anything else on the SPI bus (SD cards or touch controller). What SPI clock rate are you using?  If other devices are on the bus, what logic output state are the chip selects of other SPI devices in when the tft init function is called, are those devices cpnfigured first?

Bodmer, 2018-02-22:
Also you will need to comment out the #define M5STACK as that temporary work-around failed.

maxSnyder, 2018-02-23:
Answers to your questions:
**_•	Do you have anything else on the SPI bus (SD cards or touch controller)?_**
The SPI bus is connected to TFT_MOSI, TFT_MISO, TFT_SCK and in parallel to T_DI, T_DO, T_SCK of the XPT2046 touch screen controller on the display board. There are no other devices on the bus.
**_•	What SPI clock rate are you using?_**
The selected SPI clock rate in User_Setup.h is 40 MHz for the TFT:
#define SPI_FREQUENCY  40000000 // Maximum to use SPIFFS
and 2.5 MHz for the touch screen controller:
#define SPI_TOUCH_FREQUENCY  2500000
I tried the SPI clock rate 27 MHz, too, but with no difference in the result.
**_•	The  #define M5STACK  is commented out_**

Bodmer, 2018-02-23:
Which pins are you using for TFT_CS and the touch T_CS?  Is T_CS also defined in the setup file?

maxSnyder, 2018-02-24:
My definitions in _User_Setup.h_:
#define   TFT_CS     UNO_D10    // (= GPIO15 = SS) TFT Chip Select control pin 
#define   TFT_DC     UNO_D8     // (= GPIO0) Data/Command control pin 
#define   TFT_RST    UNO_D9     // (= GPIO2) Reset pin

#define   TOUCH_CS   UNO_D4     // (= GPIO4 = SDA) Chip select pin (T_CS) of touch screen

UNO_xx is defined in a separate header file, which I include in the User_Setup.h:
#include <wemos_D1_R2_header_pins.h>  // Assignment of Wemos D1 R2 Header Pins and esp8266's GPIOs

This is the content of wemos_D1_R2_header_pins.h:
/*
 wemos_D1_R2_header_pins.h
 Assignment of Wemos D1 R2 Header Pins (board in Arduino UNO layout) 
 and the esp8266's GPIOs
 */

#ifndef WEMOS_D1_R2_HEADER_PINS_H_INCLUDED
#define WEMOS_D1_R2_HEADER_PINS_H_INCLUDED

// UNO header pin and corresponding GPIO of the esp8266
#define UNO_D0    3  // Rx
#define UNO_D1    1  // Tx
#define UNO_D2   16  // -
#define UNO_D3    5  // SCL
#define UNO_D4    4  // SDA
#define UNO_D5   14  // SCK
#define UNO_D6   12  // MISO
#define UNO_D7   13  // MOSI

#define UNO_D8    0  // -
#define UNO_D9    2  // Tx1
#define UNO_D10  15  // SS
#define UNO_D11  13  // MOSI
#define UNO_D12  12  // MISO
#define UNO_D13  14  // SCK
//#define UNO_D14      // (GND)
//#define UNO_D15      // (n.c.)
#define UNO_D16   4  // SDA
#define UNO_D17   5  // SCL

#endif

#define SUPPORT_TRANSACTIONS
was not uncommented in User_Setup.h until now (I didn’t know, that I have to do so). 

Now, of course, I uncommented it, but the result is still the same: 
The x, y – coordinates, the touch controller delivers, appear at a different (mirrored) place on the screen and not there, where the screen was touched (I tested it with your "ON-OFF button" sketch).

maxSnyder, 2018-02-24:
My hardware connections:
[image]

Bodmer, 2018-02-24:
I am wondering if I have misunderstood what your original post said, viz "_**In all but one rotation settings 0 … 7, the screen content is mirrored**_".  Do you mean that test is written backwards, as it would be seen in a mirror, or is it just that the touch positions do not correspond with the graphics positions?

maxSnyder, 2018-02-26:
When I wrote my original post, both was the case: In all but one rotation settings 0 … 7, the screen content was mirrored, i.e. text was written backwards, as it would be seen in a mirror **_and_** the touch positions did not correspond with the graphics positions.
But now when I run the attached sketch, in all 4 rotation settings 0 … 3, there is readable text in horizontal or vertical orientation as the following pictures show, whereas only in the rotation settings 4 … 7 text is written backwards, even though I have no explanation for this changed behaviour.
But as the following pictures show, the touch positions still do not correspond with the graphics positions.
In the pictures the bigger dots named Tx are the places, where the screen was touched and the smaller dots named Sx are the corresponding circles drawn by the sketch. As one can see, they are mirrored along the dashed drawn axes.
I don't know the XPT2046 touch screen controller, but could it be, that the x, y coordinates it delivers are already mirrored and therefore it needs another initialization as it gets now?
[image]

/*
 Sketch to test the touch controller result in all 4 screen rotations
 */
#include <TFT_eSPI.h>  // Hardware-specific library
#include <SPI.h>

TFT_eSPI tft = TFT_eSPI();  // Invoke custom library

void setup(void) {
  Serial.begin(115200);
  tft.init();
  tft.setRotation(0);
  tft.setTextSize(2);
  Serial.println();
}

void loop() {
  tft.setWindow(0,0,320,320);  // taken over from your sketch without knowing what it does
  tft.pushColor(TFT_BLACK, 240 * 160);  // taken over from your sketch without knowing what it does
  tft.pushColor(TFT_BLACK, 240 * 160);  // taken over from your sketch without knowing what it does

  tft.setTextColor(TFT_WHITE, TFT_BLACK);
  tft.setCursor(0,0);
  tft.print("Rotation = "); tft.println(tft.getRotation());

  bool waiting = true;
  while(waiting) {
    while(Serial.available() == 0) {
      yield();
      uint16_t x = 0, y = 0; // To store the touch coordinates
      // pressed will be set true if there is a valid touch on the screen
      boolean pressed = tft.getTouch(&x, &y);
      // Draw a white spot at the detected coordinates
      if (pressed) {
        tft.fillCircle(x, y, 2, TFT_WHITE);
        Serial.print("x,y = ");
        Serial.print(x);
        Serial.print("/");
        Serial.println(y);
      }
    }
    unsigned int n = Serial.parseInt();
    if (n > 7) n %= 8;
    tft.setRotation(n);
    waiting = false;
  }
}

maxSnyder, 2018-02-27:
I downloaded and installed the library again. When running the test from my last comment, nothing has changed, the touch positions are still mirrored as shown in my last comment.

Bodmer, 2018-02-27:
You need to run the calibration sketch "Touch_calibrate" and include the calibration settings reported in the serial monitor in your new sketch within setup().  For example lines like:

  uint16_t calData[5] = { 286, 3534, 283, 3600, 6 };
  tft.setTouch(calData);

maxSnyder, 2018-02-28:
I run the calibration sketch "Touch_calibrate" for all 4 rotation settings and got these slightly different results:

  uint16_t calData_0[5] = { 466, 2924, 464, 3132, 1 };  // Calibration Data for Rotation 0
  uint16_t calData_1[5] = { 410, 2935, 289, 3247, 1 };  // Calibration Data for Rotation 1
  uint16_t calData_2[5] = { 471, 2980, 306, 3327, 4 };  // Calibration Data for Rotation 2
  uint16_t calData_3[5] = { 433, 2957, 286, 3302, 7 };  // Calibration Data for Rotation 3

As the corresponding first four values are nearly equal, I guess the differences can be explained by inaccuracies when touching the corners of the screen (the lower right edge of my screen is almost insensitive). The significance of the fifth value is unknown to me.

**Using these calibration values, the indicated hits are shown (almost) there, where I touch the screen!!!**

Does this practically mean that I have to store these values and to include this calibration setting in every sketch, which uses the touch function? And if so, is one set of calibration data enough for all 4 rotations (maybe using the average of the 4 corresponding values)? And what about the fifth value?
I’m very happy, that it works now fine and I thank you very much for spending your time and for your patience.

Bodmer, 2018-02-28:
The touch screen is calibrated using the Touch_calibrate sketch in screen rotation 1. The calibration values are then included in setup() in your own sketch.  The same calibration values are used for all screen rotations, the library automatically rotations the coordinate frame according to the rotation set in your own sketch.

The [Touch_calibrate comments](https://github.com/Bodmer/TFT_eSPI/blob/master/examples/Generic/Touch_calibrate/Touch_calibrate.ino#L28-L33) tell you what you need to do.

closed 2018-03-04 by Bodmer

GalBrandwine, 2019-03-17:
Hey @maxSnyder ,
Iv'e had the same problem: mirrored content on my ESP32 DEVKIT_V1 with TFT LCD Display Shield Touch Panel ILI9341 240X320.

I've played with some parameters in the tft.setRotation function and iv'e found a solution that worked for me.

// Rotation params:
// 0 - 0 deg
// 1 - 90 deg anti-clockwise (from 0 deg)
// 2 - 180 deg anti-clockwise
// 3 - 270 deg anti-clockwise
// 4 - mirror content, and rotate 180 deg anti-clockwise
// 5 - mirror content, and rotate 270 deg anti-clockwise
// 6 - mirror content, and rotate 0 deg anti-clockwise
// 7 - mirror content, and rotate 90 deg anti-clockwise
`tft.setRotation(6);`

hope it helps, Gal