Bodmer/TFT_eSPI #99: Problems with Font 8 (FONT72RLE) and background color on ST7735 1.8" 128x160 SPI screen

robcazzaro, 2018-02-26:
Try running the following code on a 128x160 TFT with ST7735 chip (mine is using #define ST7735_BLACKTAB ). It's adapted from the 128x160 RLE_font_test sample provided

#include <TFT_eSPI.h> // Graphics and font library for ST7735 driver chip
#include <SPI.h>

TFT_eSPI tft = TFT_eSPI();  // Invoke library, pins defined in User_Setup.h

void setup(void) {
  tft.init();
  tft.setRotation(1);
}

void loop() {
  tft.fillScreen(TFT_BLUE);
  tft.setTextColor(TFT_YELLOW, TFT_BLUE);

  tft.drawString("123", 0, 0, 8);
  delay(3000);
  
  tft.fillScreen(TFT_BLUE);
  tft.setTextColor(TFT_YELLOW);

  tft.drawString("123", 0, 0, 8);
  delay(3000);
}

The first tft.drawstring shows 12 followed by a corrupted (unreadable) 3. The second tft.drawstring shows 123. Please note that on a 160 pixel wide screen there isn't enough space for, 3 characters of 55 pixels each (165 pixels total), so it must be a side-effect of rolling off the screen.  When setting the starting point as tft.drawString("123", -2, 0, 8); both the first and last character get truncated in the first case, not truncated in the second (character background not set)

tft.print() has the same behavior as tft.drawString()

Not sure if it's a bug, but methinks that even if the characters roll off screen, partially truncating a character is preferable to garbled output

Bodmer, 2018-02-26:
Boundary checks have been omitted deliberately to maximise rendering speed for the usual case where the characters fit within the screen area. So the decision was made that rendering speed was more important than coping with partially off screen characters, as that case could be avoided in the sketch. The font rendering code was originally created for lower performance processors such as the UNO where the boundary checks slowed rendering significantly.

One option is to use a sprite of suitable size (maybe 80x170 pixels in 8 bit depth), draw the numbers into that and push the sprite to the screen. Sprites do get cropped to the screen boundary and negatice offsets are allowed.

closed 2018-02-26 by Bodmer

robcazzaro, 2018-02-26:
That's what I thought, hopefully will also help others by seeing the reply (thanks for closing it). In my case it's not a problem, since I can use transparent background and the character itself never gets outside the screen  (just the bounding block with "empty" space).