Bodmer/TFT_eSPI #100: How to load/use a custom RLE font?

robcazzaro, 2018-02-28:
I just created a narrow version of a large font (using Font72rle as a starting point) for my 128x160 pixel TFT

I can replace the original font72rle as font8, and use it in place of the original one, but for future compatibility reasons and library portability, I would prefer to load it as a custom font.

I do I load a custom RLE font and use it in a tft.setCursor(0, 0, 8); statement (or similar)? 

The Custom Font examples only deal with the GFXFF fonts, but I want to use RLE for memory and speed reasons

It would be good to have functionality to load custom RLE fonts, at least one at a time (it might already exist, btw, I simply cannot find it)

robcazzaro, 2018-02-28:
P.S. happy to share the font, if you like. It's just a re-encoding of your font, but having only 53 pixels width (instead of 55), it allows the display of 3 numerals on a 160 pixel screen with no truncation and fast repaint, good for a very visible counter on a small screen

Bodmer, 2018-02-28:
As you surmised, adding RLE fonts required changes to the library.  If you post the font files I will add them as a selectable alternative to the existing font 8, e.g:

#define LOAD_FONT8N

This seems a reasonable approach as the font is not sufficiently different.

robcazzaro, 2018-03-01:
I like the idea of having an option to choose the narrower 72RLE when using the 160 screen.

Ok, here are the font files (I didn't change the name). Numerals are 53 pixels wide, the 3 punctuation symbols unchanged at 29 pixels. I tested all characters, and verified they fit properly in a 160 pixels screen even when using the background color (which allows for fast repaint of text with no flickering: the new characters fully overwrite the old). The reduced spacing does not impact readability at all, even when using 4, the widest character in the set

Please note that I "moved up" all characters by 2 scan lines. Your original font is using a grid height of 75 pixels, with 4 empty scan lines at the top, and only 1 or 2 at the bottom. Since I'm drawing my counter at the top of the screen, I "moved up" the character inside the 75 pixel high grid by 2 pixels (there are now 2 empty scan lines at the top, 3 or 4 at the bottom). I don't know if I'm supposed to change anything (#define baseline_f72 73?) in the .h file.... I think that should be 71, but I'm not sure how that works (and I can't see a difference when I tested)

As a side note, quite puzzling I can't add a .c or .h file here :)... I added a .txt at the end to work around it
[Font72rle.c.txt](https://github.com/Bodmer/TFT_eSPI/files/1769614/Font72rle.c.txt)
[Font72rle.h.txt](https://github.com/Bodmer/TFT_eSPI/files/1769615/Font72rle.h.txt)

robcazzaro, 2018-03-01:
BTW: if you prefer, I can re-encode without the 2 scan lines shift, to be 100 compatible with the standard font... I sent you the file I'm using, but since I wrote a hacky utility to re-encode fonts, might as well re-run it  to have identical fonts, just narrower

robcazzaro, 2018-03-03:
Here's the version of the narrow font with the same baseline as the original font. Please use this one, since it's 100% compatible with the one currently in the build (just 53 wide instead of 55). There's no reason why anyone else would want to use the up-shifted version I'm using.
[Font72rle_53.zip](https://github.com/Bodmer/TFT_eSPI/files/1777047/Font72rle_53.zip)

Bodmer, 2018-03-04:
OK, thanks. I am adding some new library features at the and will build in the new font option for the next release.

Bodmer, 2018-03-05:
I have added the font and it is working OK, I will update the Github repository when I have tested the latest changes that support UNO format displays on an UNO format ESP32 board.

closed 2018-03-05 by Bodmer