Bodmer/TFT_eSPI #102: Convert Jpeg to C header file

danbicks, 2018-03-01:
Hi Bodmer,

What software application can I use to convert a JPG for full resolution to a C header file as per your example eagleeye in TFT_flash_jpg?

I have a couple of images I would like to convert like this rather than store them in Spiffs.

Big thanks

Dans

Bodmer, 2018-03-01:
You can use the function included, but commented out, in this sketch:
[https://github.com/Bodmer/TFT_eSPI/blob/master/examples/160%20x%20128/TFT_SPIFFS_Jpeg/TFT_SPIFFS_Jpeg.ino#L119-L126](https://github.com/Bodmer/TFT_eSPI/blob/master/examples/160%20x%20128/TFT_SPIFFS_Jpeg/TFT_SPIFFS_Jpeg.ino#L119-L126)
Store the image to convert in SPIFFS, then create the C file.

danbicks, 2018-03-02:
Thank you buddy much appreciated.

Awesome library.
Dans

closed 2018-03-02 by Bodmer