Bodmer/TFT_eSPI #103: Text rotation

danbicks, 2018-03-04:
Hi Bodmer,

I have my display in landscape mode and have drawn a horizontal bar graph vu meter all works brilliant. I have now created a similar routine for vertical on a landscape theme but can't seem to find a method to rotate the text message. Any ideas?

Big thanks

Dans

Bodmer, 2018-03-05:
Text can only be printed horizontally in any screen mode.

One option is to create a bitmap of the text on a PC and load that into SPIFFS, then render to screen.

I will think about supporting rotating text but I am busy on adding parallel interface support to the library at the moment.

closed 2018-03-05 by Bodmer

Gun-neR, 2020-06-03:
Hello.  Did this option get introduced?  I havent yet found any commands to indicate that it has, thus the question.