Bodmer/TFT_eSPI #104: Compiler error: SPI not found when touch activated? [ESP32] [ili9341] [Windows]

The1Rob, 2018-03-12:
Hey Everyone,
First off, I want to give a HUGE thank you to this library and community.  I was having a hell of a time getting displays working with the ESP32 until I found this library.

I am having one issue though.  Once I uncomment the TOUCH_CS line in user_setup, and uncomment support_transactions, when i compile I get a "SPI method not found" error. If i comment out the touch_cs line, everything compiles and works fine (woohoo!)

I'm using Arduino 1.8.5, everything up to date.  esp-idf installed, esp core for arduino loaded.  Wifi and display tests and work flawlessly.  

It's a ESP32-devkit board, and a 2.8" 320x240 touch screen using ili9341.  Sketch is the example Touch_calibrate.

Am I missing something?  Below is the error.  Thanks in advance.  =)

*snip*

`In file included from C:\Users\robertk\Documents\Arduino\libraries\TFT_eSPI\TFT_eSPI.cpp:16:0:

C:\Users\robertk\Documents\Arduino\libraries\TFT_eSPI\Extensions/Touch.cpp: In member function 'uint16_t TFT_eSPI::getTouchRawZ()':

C:\Users\robertk\Documents\Arduino\libraries\TFT_eSPI\TFT_eSPI.h:203:27: error: 'class SPIClass' has no member named 'SPI'

     #define transfer16(C) SPI.write16(C)

                           ^

C:\Users\robertk\Documents\Arduino\libraries\TFT_eSPI\Extensions/Touch.cpp:58:13: note: in expansion of macro 'transfer16'

   tz += SPI.transfer16(0xc1) >> 3;

             ^

C:\Users\robertk\Documents\Arduino\libraries\TFT_eSPI\TFT_eSPI.h:203:27: error: 'class SPIClass' has no member named 'SPI'

     #define transfer16(C) SPI.write16(C)

                           ^

C:\Users\robertk\Documents\Arduino\libraries\TFT_eSPI\Extensions/Touch.cpp:59:13: note: in expansion of macro 'transfer16'

   tz -= SPI.transfer16(0x91) >> 3;

             ^

Using library SPI at version 1.0 in folder: C:\Users\robertk\Documents\Arduino\hardware\espressif\esp32\libraries\SPI 
Using library TFT_eSPI at version 0.19.10 in folder: C:\Users\robertk\Documents\Arduino\libraries\TFT_eSPI 
Using library FS at version 1.0 in folder: C:\Users\robertk\Documents\Arduino\hardware\espressif\esp32\libraries\FS 
Using library SPIFFS at version 1.0 in folder: C:\Users\robertk\Documents\Arduino\hardware\espressif\esp32\libraries\SPIFFS 
exit status 1
Error compiling for board ESP32 Dev Module.`

Bodmer, 2018-03-12:
OK, thanks for pointing this out, I seem to have broken the touch support in the last update. I will correct this within 24 hours.

Bodmer, 2018-03-12:
I have fixed this in the Github copy. This was a name conflict between the transfer16() macro and the SPI library function.

closed 2018-03-12 by Bodmer