Bodmer/TFT_eSPI #105: Support for eInk ePaper displays?

JamesNewton, 2018-03-15:
Sorry to raise this as an issue, but I have no other way to contact you.

Would you be interested in receiving a couple eInk displays to see if this library could be updated to work with them? We are willing to ship you a pair in the UK if you would perhaps be interested in adding support for them via the overlap mode (need overlap as there are so few pins on an ESP).

We are looking at this display:
https://www.waveshare.com/1.54inch-e-Paper-Module.htm
and also at this display:
https://www.waveshare.com/1.54inch-e-Paper-Module-B.htm

But that second one, the red/black/white, is rather difficult to work with. 13 seconds to update, and you must build up the entire bitmap in memory before sending it all out in one shot.

The first one, black and white only, is a bit easier as you can set up an area, and just send a bitmap for that area alone. 

Of course, we would prefer the color, and we only really care about text, but of course you get the one with the other. Honestly, we appreciate your work on this library so much that you are welcome to the displays even if you don't end up adding support. eMail me an address to JamesNewton@MassMind.org

Bodmer, 2018-03-15:
As it happens I already have a 2.7" WaveShare ePaper display but have not gotten around to looking at the specs for driving it.

I suggest I try to get the one I have running before any hardware is sent! I will email you.

closed 2018-03-15 by Bodmer

Bodmer, 2018-03-15:
I have connected the 2.7" ePaper display and adapted the WaveShare demo code to work with an ESP32 and it runs fine.  I will look at it further over the next few days and see how the code can be added to the TFT library, it looks quite straightforward, but I have only scanned the files briefly!

Bodmer, 2018-03-17:
Initial results are quite promising, the code is still a work in progress.  I have experimented and used a Sprite as a 1 bit per pixel screen buffer, the graphics can then be drawn into the sprite and the image passed to the existing Waveshare library for display. This make supporting multiple sizes and 2 or 3 colour ePaper screens relatively straightforward and the complexity of image transport to the ePaper display is then hidden and needs no special support in the TFT_eSPI library.

[https://i.imgur.com/L2tV129h.jpg?1](url)

JamesNewton, 2018-03-19:
We are happy to buy you a 3 color (black / white / read) display if you would like one. Keep in mind they can NOT be updated in part. It's the entire display or nothing, so there is a significant use of RAM for the 3 color units. I haven't looked closely at the datasheet, but I wonder if you could 1. start the display update, 2. Render the first X lines of the display into a sprite, 3. send that data, then repeat steps 2 and 3 until the entire display had been drawn. It would require some sort of loop or call back in the program to render each line of text (assuming you wanted to display text) each time the screen is updated. Still, an array containing all the lines of text, and then a small bit of sprite ram to render each line in turn is far less than would be needed to render the entire display. 

Anyway, great news!

Bodmer, 2018-03-20:
The RAM should not be an issue as each full frame buffer only consumes 5kbytes = 10kbytes total and typical sketches have >20kbytes free.  As far as I can see you can also just maintain one screen buffer, fill with black graphics and send to display, fill with red graphics and send to display and the tell the display to refresh.

I have a temporary library version ready, do you have displays to test the library on?

JamesNewton, 2018-03-20:
We actually dont have one of these on hand. Can we place an order and drop ship a 3 color unit to you?

Bodmer, 2018-03-20:
I will contact via email...