Bodmer/TFT_eSPI #106: Got 'class TFT_eSPI' has no member named 'getTouch' on ESP32/ILI9341 Parallel

SandUhrGucker, 2018-03-16:
Hi,
great work so far! Graphics wors like a charm, but  I got 'class TFT_eSPI' has no member named 'getTouch' in Keypad Example and others who use the Touch. 

Also I miss a Donation Button ;-)

Bodmer, 2018-03-16:
The touch controller functions are currently disabled for parallel interface displays.

The touch support built into the library expects an SPI based XPT2046 controller. Most UNO style parallel displays deem to require analogue sampling.  Adding touch support for UNO style displays may be possible but I have not tried it yet, it is on my "to do" list, so I will leave this issue open for now.

What type of display touch controller is used on your display?

Mmmm, I'll think about adding a donation button!

SandUhrGucker, 2018-03-16:
Hi,
I dont know the touch-controllertype is used. But:
The Display is an original mcufriends 2,4" Display, wich runs perfectly stacked on an Arduino Uno by using the mcufriends library (https://github.com/prenticedavid/MCUFRIEND_kbv).
The ReadID Script identify the Device as a 0x9302.
The Pin-Diagnostic Script on the Uno says it is using A1, A2, D6, D7.
Here is a Picture of the Display: https://picload.org/view/dalwpidi/2.4_lcd_tft_arduinor3_bild3.jpg.html 
Here is a Picture of the Back: https://picload.org/view/dalwpidw/2.4_lcd_tft_arduinor3_bild2.jpg.html

It would be really phantastic, if you were able to implement the touch in your library! - Be sure I will push your Button ;-)

Bodmer, 2018-03-17:
The screen looks like the common type with resistance based touch screen, these require analogue reads of pins and this if not supported yet.  I have a couple of screens like that so I will experiment with them, but this may take a while as I am currently experimenting with adding support for ePaper screens.

closed 2018-03-25 by Bodmer