Bodmer/TFT_eSPI #107: can fonts from squix be used

al-ang, 2018-03-17:
I'm using your weather-station-v8 example but want to use the moonphases.h fonts from squix's latest.  So far I have not been able to get the font working.  Sorry for asking something that's probably simple, but how can I get the moonphases.h font to work with your TFT-eSPI library?

Bodmer, 2018-03-17:
Daniel Eichhorn (Squix) has completely re-written the graphics library and invented a new font format which i (as you found) not compatible with the TFT_eSPI library :-(

I will have a look and see if it is possible to adapt the code.

Bodmer, 2018-03-22:
Supporting the different font format would need some significant changes, so this is some way down on the "To Do" list.

closed 2018-03-22 by Bodmer

al-ang, 2018-03-23:
I appreciate your taking time to look into the issue