Bodmer/TFT_eSPI #109: 2 displays

monospacesoftware, 2018-03-18:
I can't find a way to get this library to talk to 2 displays at the same time.  Is this possible? The usage of #define for TFT_CS and TFT_DC, rather than these being passed as constructor parameters, appears to limit the library to 1 display only.  I wire up 2 with shared MOSI, MISO, and SCLK pins but separate CS and DC pins.

Bodmer, 2018-03-20:
What I have done in the past is connect all lines to both displays except have separate pins for the two chip selects. Then:
1. Do not define the chip select pin in the library.
2. Only create one instance
3. Control the chip selects in your sketch depending on which display you wish to control.

closed 2018-03-25 by Bodmer

alwan12e3, 2019-11-13:
Im working on the similar project now.

How can we actually control the chip select?