Bodmer/TFT_eSPI #112: The Button class from Adafruit_GFX example keypad

dragospascale, 2018-03-22:
if tft.setRotation(1);
touch coordinates are not rotated and therefor buttons respond to touches outside their boundaries.
You can check this in your own example keypad_480x320 where you set tft.setRotation(1);

Bodmer, 2018-03-22:
Run the example/Generic/Touch_calibrate sketch.

Do the white spots appear at the touch positions once calibration is complete?

dragospascale, 2018-03-23:
No. The x and y axes seem to be reversed.

dragospascale, 2018-03-23:
I'm using ILI9431 with XPT2040 320x240 display. The problem is with touch. Graphics are ok.

dragospascale, 2018-03-23:
In Touch.cpp I've forced touchCalibration_invert_x = false; and touchCalibration_invert_y = true;
The if X test returned true and if Y test returned false. Now everything is ok but it is not an elegant solution.

Bodmer, 2018-03-23:
Are you using a Raspberry Pi display, if not can you provide a web link to the item?

dragospascale, 2018-03-26:
https://www.aliexpress.com/item/2-8-inch-240-320-HD-color-display-SPI-with-touch-panel-TFT-LCD-module-11pins/32795001104.html

Bodmer, 2018-03-28:
OK, thanks for the link, I guessed it was a display that I have not tested!

I suspect the touch rotation code is not handling all the 8 possible TFT to touch coordinate conversions needed. I will have to investigate further.

Bodmer, 2018-04-01:
I have reversed the coordinate transform, that should fix it for your TFT.

I will close this but I have raised Issue #115 to investigate further as I think the transform does not cater for all possible orientations.

closed 2018-04-01 by Bodmer