Bodmer/TFT_eSPI #113: Wouldn't It Be Nice

dragospascale, 2018-03-26:
First, thank you for your great work! I hope you don't mind if I'll make you a few suggestions for future development. Maybe I don't understand your passion for sprites. I don't really understand their utility. In exchange, I'll be very happy if you can develop more useful controls like spinners or drop-down fields or text fields with editing facilities. It will be very useful to have for example a field for setting time values. You know better how to treat touch events.  Do you think it's possible to develop a way, (based on pressure maybe) to repeat the same command if I keep touching the same control more than a few seconds? I think that for home automation these kinds of controls will be more than useful. I hope that you will receive this message just like a suggestion for your future work and again I'm so grateful for such a useful code you provided to the community. Thank you again.

Bodmer, 2018-03-28:
Requested features are always welcome but I am time constrained for implementing them!

The passion for Sprites came during the coding while they were added and the realisation of what becomes possible and was messy to handle before was messy to do before, like flicker free updates of changing graphics. As it happens they would enable drop down menus and "spinners" however the "touch" experience with the cheap resistive touch screen leaves a lot to be desired and that constrains the GUI somewhat.

There have been a few attempts at coding GUIs but their usability is somewhat lacking and not very attractive.

I use the TFT's for information display only usually and thus have not had a need for a GUI and send the data to an Android phone, or or use real physical buttons for controls, That way the ESP8266 connected screen remains finger smear and scratch free.

mgx0, 2018-03-30:
Hello,

When I replace the text (update value) on a TFT, i use to fill the frame with background color and draw the string again with updated value. 
Is there a way to prevent flicker by building the layout without displaying it on the TFT and display it once the layout is ready?

Imagine having 10 values changing every second, pretty flickering display 😁

Bodmer, 2018-03-31:
@mgx0  Yes, you have the answer!

closed 2018-03-31 by Bodmer

Bodmer, 2018-03-31:
I have ordered a few "APDS-9960 RGB and Gesture Sensor" boards that are available at low cost online. It seems they would be a neat way of controlling the information displayed on a TFT. It would mean the screen does not have to be manually picked up to use the touch screen. I will experiment when they arrive.