Bodmer/TFT_eSPI #114: No touch function in de latest version

Nulista, 2018-03-30:
Hello Bodmer,

You removed de touch function from the latest version.
Why? 

Thx
Ludo

Bodmer, 2018-03-30:
Oh! That was not intentional, I will investigate.
Thanks for reporting this.

Bodmer, 2018-03-30:
Incidentaly, the touch functions have been moved to the "Extensions" folder, in case you were looking for them!

Nulista, 2018-03-30:
thanks, but now i tested the library with the ESP32, and the TFT did not work, example "meters"
I don't no wat is wrong..??

Bodmer, 2018-04-01:
Do the other examples work?
Have you configured the User_Setup.h file for the pins you use. If you used the User_Setup_Select.h then which setup have you chosen.

What display do you have (size, driver etc)?

Nulista, 2018-04-01:
Hello,
Yes i have configured the User_Setup.h file. I use the RPi 3.5 SPI LCD wih the ESP32.
And it is with all the examples.
With the older version it works fine, but with the latest version, there is something wrong...
No properly reset,  perhaps there is something with the pins allocation in the program??

I hope that you could help me?
Thanks

Bodmer, 2018-04-01:
I will try connecting an ESP32 to my RPi TFT and see what is wrong. I will post back soon.

Meanwhile, can you temporarily revert back to the older wotking TFT_eSPI?

Bodmer, 2018-04-01:
Hi,

I have fixed part of the problem, the Git repository has been updated.  The RPi TFT now works with the ESP32.

I have not looked at the Touch issue yet.

Bodmer, 2018-04-01:
Touch is also fixed.

Thanks for reporting this. If you still have problems please raise a new Issue.

closed 2018-04-01 by Bodmer

Nulista, 2018-04-01:
Hello,

 

Indeed, all is working fine now! 

Thx a lot!

 

Mvg

Ludo

 

Closed #114 <https://github.com/Bodmer/TFT_eSPI/issues/114> .