Bodmer/TFT_eSPI #116: How to reduce flickering?

ferazambuja, 2018-04-01:
Sorry, not an issue. First great library, love the smooth fonts.

Is there any way to reduce the flickering when displaying new values? Is adding a delay the only option?
I'm having trouble to find material about it.
 
Thanks

Bodmer, 2018-04-02:
There is a function to [printToSprite() ](https://github.com/Bodmer/TFT_eSPI/blob/master/Extensions/Sprite.cpp#L1493)in the library but it is has not been extensively at the moment, so you may find a bug!

If a Sprite does not exist it creates one, renders the text to the screen at the cursor coordinates and deletes the sprite to recover memory.  I have not created an example yet either.

Bodmer, 2018-04-02:
If you experiment, beware that the printToSprite() function creates a sprite that is big enough to hold the entire string, so make sure you have enough RAM spare!

ferazambuja, 2018-04-02:
Thanks
Going to check out 👍

NdK73, 2018-04-04:
I prefer avoiding printToSprite(), declaring explicitely the sprite dimensions.
This is a snippet of the code I'm using:
```
void GUI::setMessage(const char *msg) {
  TFT_eSprite m = TFT_eSprite(&disp);
  m.setColorDepth(8);
  m.createSprite(disp.width(), 24);
  m.fillSprite(TFT_BLACK);

  m.setTextFont(1);
  m.setTextDatum(TL_DATUM);
  m.setTextColor(TFT_WHITE);
  m.setCursor(0, 0);
  m.print(msg);
 
  m.pushSprite(0, MSG_YPOS);
  m.deleteSprite();
}
```

Bodmer, 2018-04-04:
OK, thanks, that will give you better control of the sprite width if you want to over-write old text.

closed 2018-04-05 by Bodmer