Bodmer/TFT_eSPI #117: 'class TFT_eSPI' has no member named 'getTouch'

al-ang, 2018-04-04:
Wanted to explore the TFT_eSPI library touchscreen capabilities and started with Keypad_240x320 example.  Sketch would not compile:
```

Keypad_240x320:111: error: 'class TFT_eSPI' has no member named 'getTouch'
   boolean pressed = tft.getTouch(&t_x, &t_y);
```

al-ang, 2018-04-04:
Thank you. the sketch compiles now.  I appreciate your help, as well as the library you've shared.

Editing the library feels weird to me.  I use different hardware, e.g. nodemcu, wemos d1 mini, nano v3, and soon esp32.

I *think* TFT_eSPI is setup so I can edit each section for different hw, and each section is picked up during compile by directive flags, but I haven't confirmed that yet.  Up to now I've been comfortable making defines inside each sketch, making hw dependency changes in the sketch. I like to minimize library changes since it can impact so many sketches, vs modifying individual sketches as necessary.

Regardless, I appreciate the help. I'm having a lot of fun with the library.

closed 2018-04-04 by al-ang

Bodmer, 2018-04-06:
You have to select just one setup in the User_Setup_Select.h file.  This arrangement is not ideal but the Arduino IDE does not let the library see #defines in the sketch and for testing reasons with many different connection setups I find it is a pain having multiple copies of sketches for different display+processor+connection setups.

The key advantage for me therefore is that once you have edited the User_Setup and tailored it for your needs then all sketches should run fine without making any changes to them.  The alternative is for users to edit every example sketch they want to test.

If you are using PlatformIO then see the note in the Tools folder of the library.

Wolfgang-Wulz, 2022-09-07:
Hello and before any complain, thank you for a great library I really enjoy to use.
But now, I have same problem about the getTouch/setTouch.
Using an ESP32 WROOM and TFT ILI9341 and there is no problem about display and read from SD.Card and what so ever. 
Just now, I wanted to integrate the touch function and it tells me same problems as the guy before.

User_Setup.h edited like
#define TOUCH_CS 5     // Each Chip-Select (CS) needs extra GPIO

no matter which setting I tried, result is seen below

The message from compiler
...\TFT_eSPI\examples\480 x 320\Touch_Controller_Demo\Touch_Controller_Demo.ino: In function 'void setup()':
Touch_Controller_Demo:43:9: error: 'class TFT_eSPI' has no member named 'setTouch'
     tft.setTouch(calibrationData);
         ^
Touch_Controller_Demo:46:9: error: 'class TFT_eSPI' has no member named 'calibrateTouch'
     tft.calibrateTouch(calibrationData, TFT_WHITE, TFT_RED, 15);
         ^
...\TFT_eSPI\examples\480 x 320\Touch_Controller_Demo\Touch_Controller_Demo.ino: In function 'void loop()':
Touch_Controller_Demo:63:11: error: 'class TFT_eSPI' has no member named 'getTouch'
   if (tft.getTouch(&x, &y)) {
           ^
exit status 1
'class TFT_eSPI' has no member named 'setTouch'

Thanks in advance

Wolfgang-Wulz, 2022-09-08:
Thanks a lot. As usual in such cases, the error was sitting in front of the computer!

summerfind, 2023-06-19:
> Run the Read_User_Setup example and check the touch pin is listed in the serial output.

Hi Bodmer, 
seems this code can read User_Setup information from file? if so, whats the use of? 
do you have a example that can read the real wiring information? 
Thanks

tochinet, 2024-01-22:
> > You have to select just one setup in the User_Setup_Select.h file. This arrangement is not ideal but the Arduino IDE does not let the library see #defines in the sketch and for testing reasons with many different connection setups I find it is a pain having multiple copies of sketches for different display+processor+connection setups.
> > The key advantage for me therefore is that once you have edited the User_Setup and tailored it for your needs then all sketches should run fine without making any changes to them. The alternative is for users to edit every example sketch they want to test.
> 
> I'm new to this so I could be completely wrong, but I would think the board definition in the variant.h would be a good place for putting a setting like this. Either that or sketch by sketch. Is there a way to edit this sketch by sketch? Is it possible to use have it in the board definition?
> 
> My terminology could be wrong here. It's board definition or platform, not sure which.

It's a never ending story. The problem of the Arduino platform is you are easily drowning in 100+ boards  you don't have, and selecting a board can't be done in the sketch for bad reasons. They prefer you to select a board and make it blink, for immediate reward, than being project based. Having to modify variant.h is even 1000X worse than modifying a library, because it's even less visible.  Using platform.io means spending again large amouints of precious time on your "toolchain" for every project, not really better. As if you needed to change the handle of your screwdriver between each screw. I really can't see the value op hiding the board specifics from a sketch. If neede, it could be in a boardinfo.h in teh sketch directory. Not in the library path. But anyway,