Bodmer/TFT_eSPI #118: TFT-sSPI Touch (no touch response)

al-ang, 2018-04-05:
Having problem getting touch to work with TFT-eSPI.  Hoping problem is obvious, but I'm stuck.  Here's what I have;  a NdodeMCU and a 2.4 TFT LCD ILI9341 display.  I installed a fresh, clean copy of the TFT-eSPI library and have made only one change identified below.   Physical connections are:
NodeMCU     ili9341 (2.4 TFT SPI)
D1                  T_IRQ
D6                  T_DO
D7                  T_DIN
D2                  T_CS
D5                  T_CLK
D6                  SODI(MISO)
3V3                 LED
D5                   SCK
D7                   SDI(MOSI)
D3                   D/C
D4                   RESET
D8                   CS
GND                GND
VCC                VCC

In TFT-eSPI User_Setup.h uncomment line 95

#define TOUCH_CS PIN_D2     // Chip select pin (T_CS) of touch screen

Copied examples/Generic/Touch_calibrate to working folder.  Compiled, loaded.

On execute the calibration screen appears, and the corner cursor shows in 
each corner and moves on automatically while nothing touching the screen.  When screen says ""Touch screen to test!" nothing happens when screen is touched.

Although it's not helpful for detail work, I'm also attaching a picture of my setup.  I also tried this on a different display and a wemos d1 mini with no luck.  Am I doing something that's embarrassingly obvious.
[image]

al-ang, 2018-04-05:
I'm wondering if I have an oddball display (got it from aliexpress).  Does your library have a tft.readID so I can verify the device compatibility?   Although I've traced the connections several times, and completely re-wired them, I still think it's possible my eye and my mind are not agreeing, so I keep rechecking.

Bodmer, 2018-04-06:
You connections look fine and the good news is that the TFT is working.

There was a bug in the touch library that I updated a few days ago that showed the symptoms you describe. Can you try a brand new copy of the library?

 I have the exact same deisplay so if you cannot get it it working post back and I willl connect up mine and test it.  The library does not use the T_IRQ pin so you do not need to connect that up, however that is not why your setup not working.

al-ang, 2018-04-06:
I re-wired (to be xtra-sure) and grabbed the latest version of the library.  Uncommented the touch_cs line in User_Setup.h and uncommented the main loop Serial.print lines in a fresh copy of Touch_calibrate. As before, it runs right thru to "touch screen to test!" and x,y = 0,0 is repeated printed in the serial monitor. It seems tft.getTouch  to be returning a valid touch every time it's called, even when there's no touch?  And no change when there is a touch.

I haven't moved to platformio yet and have arduino ide 1.8.5. I also have a fairly clean library, having reinstalled everything last week.

Bodmer, 2018-04-06:
OK, I have an identical looking display so I will test it. Based on your experience I think there must be a bug in the library. Thanks for reporting this and persevering.

Bodmer, 2018-04-06:
I have connected my display up exactly as you have and the touch works OK, but when touching the screen at the end of the test the white spot that indicates the touch position is in the wrong place.

Bodmer, 2018-04-06:
In the photo it looks like D0, D1 and D2 are not connected correctly, but as you have rewired it I would check again.

Bodmer, 2018-04-06:
I have corrected the coordinate swap and my screen works fine.

al-ang, 2018-04-06:
I'm encouraged since you got it to work.  My next step is to erase Arduino on this machine and reinstall as necessary to build the sketch.  Maybe I'm picking up something from another library ... I know, it's a stretch, but it's something I can do.

The photos are not good, but I'm including my recent config.  Shared pins have the same color, and every pin on the nodemcu has it's own color.  The theory is that it'll help isolate wiring problems.   Paralax is a problem, so I'll write the color wiring color scheme:

```
D6  T_DO     ylw (shared with MISO)
D7  T_DIN    org (shared with MOSI)
D2  T_CS     brn
D5  T_CLK    grn (shared with SCK)
D6  MISO     ylw (shared with T_DO) 
3V  LED      red (3V3 shared with VCC)
D5  SCK      grn (shared with T_CLK)
D7  MOSI     org (share with T_DIN)
D3  DC       blue
D4  RESET    pur
D8  CS       wht
G   GND      blk
3V  VCC      red  (3V3 shared with LED)
```

Thank you for checking out the config.  Since I know it'll work, I can keep working it until I find my problem.
[image]
[image]

Bodmer, 2018-04-07:
I see no errors in your wiring. I will write a touch diagnostic sketch later today to see if we can isolate the problem.

al-ang, 2018-04-07:
Thank you.  Sorry for being a pain, but I really like your library and want to use it. 

From: Bodmer 
Sent: Saturday, April 7, 2018 3:58 AM
To: Bodmer/TFT_eSPI 
Cc: al-ang ; Author 
Subject: Re: [Bodmer/TFT_eSPI] TFT-sSPI Touch (no touch response) (#118)

I see no errors in your wiring. I will write a touch diagnostic sketch later today to see if we can isolate the problem.

Bodmer, 2018-04-07:
Try the attached and let me know what happens. The sketch sends raw unprocessed coordinates to the serial port along with "z" which is the output of the touch pressure calculation. These outputs will be "noisy" as they are not filtered. Touching the screen should change the values. Post the untouched readings and also when pressing near the screen centre.  The z value should be low (<100) when not touched and increase rapidly as pressure is applied to the screen.

[Test_Touch_Controller.zip](https://github.com/Bodmer/TFT_eSPI/files/1886501/Test_Touch_Controller.zip)

al-ang, 2018-04-07:
It doesn’t seem to me that touch is responding.

Starting...
x: 2952     y: 3358     z: 5300 
x: 2946     y: 3351     z: 5326 
x: 2908     y: 3319     z: 5215 
x: 2992     y: 3423     z: 5206 
x: 2974     y: 3423     z: 5239 
x: 2959     y: 3400     z: 5302 
x: 2959     y: 3394     z: 5271 
x: 2979     y: 3423     z: 5253  < around here touched one side and slide across lengthwise
x: 2961     y: 3375     z: 5278 
x: 2983     y: 3419     z: 5267 
x: 3008     y: 3431     z: 5235 
x: 2974     y: 3416     z: 5239 
x: 2938     y: 3359     z: 5274 
x: 2970     y: 3391     z: 5260 
x: 3003     y: 3463     z: 5256 
x: 2986     y: 3417     z: 5254 
x: 3000     y: 3439     z: 5224 
x: 2996     y: 3426     z: 5255 
x: 2999     y: 3452     z: 5195 
x: 2998     y: 3439     z: 5234 
x: 2983     y: 3427     z: 5263 
x: 2942     y: 3379     z: 5308 
x: 2959     y: 3390     z: 5280 
x: 2975     y: 3427     z: 5256  <ended slide around here
x: 2947     y: 3351     z: 5295 
x: 2943     y: 3389     z: 5293 
x: 2979     y: 3415     z: 5204 
x: 2975     y: 3422     z: 5265 
x: 3007     y: 3473     z: 5202 
x: 3000     y: 3439     z: 5208 
x: 2990     y: 3435     z: 5215 
x: 2987     y: 3428     z: 5274 
x: 2995     y: 3439     z: 5202 < around here touched one side and slide across lengthwise again
x: 2993     y: 3427     z: 5244 
x: 2951     y: 3378     z: 5275 
x: 2974     y: 3411     z: 5249 
x: 2996     y: 3439     z: 5239 
x: 2982     y: 3411     z: 5255 
x: 2991     y: 3449     z: 5222 
x: 2992     y: 3415     z: 5256 
x: 2940     y: 3382     z: 5302 
x: 2952     y: 3357     z: 5303 
x: 2944     y: 3331     z: 5054 
x: 2951     y: 3378     z: 5297 
x: 2946     y: 3359     z: 5307 
x: 2961     y: 3371     z: 5289 
x: 2963     y: 3383     z: 5254 
x: 2991     y: 3450     z: 5232 
x: 2988     y: 3423     z: 5252 
x: 2963     y: 3395     z: 5254 
x: 3005     y: 3453     z: 5227 
x: 2987     y: 3437     z: 5243 
x: 2970     y: 3406     z: 5258 
x: 2992     y: 3416     z: 5257 
x: 2998     y: 3439     z: 5213 
x: 2991     y: 3448     z: 5216 
x: 2977     y: 3390     z: 5266 
x: 2991     y: 3448     z: 5251 
x: 2957     y: 3384     z: 5274 
x: 2992     y: 3416     z: 5239 <ended slide around here
x: 3007     y: 3471     z: 5194 
x: 3004     y: 3446     z: 5225 
x: 2978     y: 3407     z: 5229 
x: 2942     y: 3379     z: 5298 
x: 2960     y: 3374     z: 5306 
x: 2945     y: 3344     z: 5056 
x: 2965     y: 3393     z: 5299 

From: Bodmer 
Sent: Saturday, April 7, 2018 8:34 AM
To: Bodmer/TFT_eSPI 
Cc: al-ang ; Author 
Subject: Re: [Bodmer/TFT_eSPI] TFT-sSPI Touch (no touch response) (#118)

Try the attached and let me know what happens. The sketch sends raw unprocessed coordinates to the serial port along with "z" which is the output of the touch pressure calculation. These outputs will be "noisy" as they are not filtered. Touching the screen should change the values. Post the untouched readings and also when pressing near the screen centre. The z value should be low (<100) when not touched and increase rapidly as pressure is applied to the screen.

Test_Touch_Controller.zip

Bodmer, 2018-04-07:
It looks like either a touch screen film is not fitted to your screen (!) or there is a broken connection between the touch film and the XPT2046 controller.  There should be a ribbon coming off the display like this:
[https://i.imgur.com/e55rHuX.png](url) 

Can you provide a picture of the front of your screen?

Bodmer, 2018-04-07:
I have a 2.8" screen that I bought as not being a touch screen but it has the touch controller IC fitted. So I connected it up and I get the same readings you do:

x: 3024     y: 3482     z: 5282 
x: 3016     y: 3476     z: 5279 
x: 3003     y: 3472     z: 5256 
x: 3007     y: 3503     z: 5248 
x: 3039     y: 3541     z: 5511 

 - so this suggests the screen is not fitted or not connected.

These are readings from a good screen:

x: 416     y: 3741     z: 23 
x: 416     y: 3711     z: 21 
x: 1937     y: 2480     z: 1470  <<< touched here
x: 1952     y: 2492     z: 1774

al-ang, 2018-04-07:
You hit the nail right on the head.  During testing I've used 3 different displays thinking that one could be malfunctioning.  None of the three tested have the extra wiring on the front for touch.  Luckily I had one brand new one that has the necessary connection.  As expected it works like a charm.

I am deeply embarrassed.  This seems so obvious.  (I might have thought that having the touch pins on the back meant it was touch capable.  Now I'll look for the wiring on the front.)

Thank you for the time you've spent helping me debug this problem.

This is the front of a display WITHOUT touch connections:
[image]
This is the front of a display WITH touch connections:
[image]

The back of both displays look exactly the same [to me].

closed 2018-04-07 by al-ang

wysiecki, 2018-04-18:
unbelivable, i’ve lost almost 3 months of my live!
U save my life, gentlemen.
Have also touch without touch.

Thx

al-ang, 2018-04-18:
The learning curve was high but now I know what to look for when I buy displays if I want a touch interface.  Mr Boedner was exceptionally patient helping me "debug" this problem.  

BTW, this library rocks.  I'm now using an esp8266 to drive a 3.5 display that use to require a rasperberry pi.  RPi's are great, but sometimes an esp8266 is a much more appropriate fit for the project.

Bodmer, 2018-04-19:
Thank you for your positive feedback, that makes all the effort that has gone into this library worthwhile.

PLSousa, 2018-12-26:
Identical problem using a 2.8 TFT LCD ILI9341 display [https://fr.aliexpress.com/item/2-8-240x320-SPI-TFT-LCD-Serial-Port-Module-With-PCB-Adapter-Micro-SD-ILI9341-5V/32889801161.html?spm=a2g0w.10010108.1000016.1.63011710pkW3nA&isOrigTitle=true](url). I understan now why it doesn't work - it is a display WITHOUT touch connections. Thank for your help.

pploume, 2019-01-02:
Hi,
Same problem but the display (not exactly the same) is supposed to have a touche panel ([from here](https://www.aliexpress.com/item/3-5-inch-480-320-SPI-Serial-TFT-LCD-Module-Display-Screen-with-Touch-Panel-Driver/32908809356.html?spm=a2g0s.9042311.0.0.3da24c4dOelKaN))

I've tryied the Test_Touch_Controller.zip, 
ever the same data displayed : X:0 Y:0 Z:4095
wherever I slide my finger

I've double check connections, I've tryied with another (same model) display ... still not detecting the touch.
Is it a way to check the touch panel is working ? (even if I guess 2x won't be broken ....)

BTW : other demo codes are working well !
Thanks for the great job
... and thanks in advance for your help and suggestions

PS: link to download the docs, mostly in chinese : **http://t.cn/EwL6XZh**
p

Bodmer, 2019-01-03:
The ILI9488 TFT is unusual in that it does not tri-state the SDO line even when the CS line is high. So do not connect the TFT to the MISO line, just connect the Touch T_ DO line to the MISO pin.

There is a touch diagnostic sketch in the Test and Diagnostics examples folder.

ipqate, 2019-01-03:
Hello,

I am using an ESP32 + TFT with ILI9488 and the xpt2046. The result I get is all zeros:

[image]

Which is correct according the logic analyzer:

[image]

The schematic for the xpt2046 is:

[image]

And the user setup:

[image]

I can load and see on screen almost all graphic examples, but I'm blocked with the touch panel...any idea?

Thank you in advance.

pploume, 2019-01-03:
Hi,

As simple as that !
Thanks a lot, I won't be able to find this trick by myself.

Again, really great job ... and quick and efficient support.
Congrat'
:)

Bodmer, 2019-01-04:
@ipqate

The ILI9488 TFT is unusual in that it does not tri-state the SDO line even when the CS line is high, and the XPT2046 cannot overpower the TFT output. So do not connect the TFT to the MISO line, just connect the Touch T_ DO line to the MISO pin.

ipqate, 2019-01-04:
I was worried about removing the miso because I did not know whether the screen was going to keep on working and I had to cut some tracks on a custom pcb.

But after all the stress, everything is working:)

Thank you for your support and your great library:)

Bodmer, 2019-01-04:
Glad to hear it is working.

The TFT MISO line is only needed if you want to read back pixel values from the screen. It took me a while to figure out what was happening. An oscilloscope showed the bus conflict on the line. If you wish to read from the display a tristate beffer would be needed. I expect just a 1K series resistor to the TFT SDO pin would also work so that the XPT2046 can still drive the line, but you may need to reduce the SPI read clock frequency in that case (there is a setup option for this) as the resistor will slow the edge speed on the data line due to parasitic capacitance.

FullMetalRico, 2019-08-16:
> You hit the nail right on the head. During testing I've used 3 different displays thinking that one could be malfunctioning. None of the three tested have the extra wiring on the front for touch. Luckily I had one brand new one that has the necessary connection. As expected it works like a charm.

I just wanted to open an Issue because i am getting kind of the same results as those ones:
> x: 3024 y: 3482 z: 5282
x: 3016 y: 3476 z: 5279
x: 3003 y: 3472 z: 5256
x: 3007 y: 3503 z: 5248
x: 3039 y: 3541 z: 5511

and could not find an open issue with that but you just saved me a whole ton of time with that. Really sad that it has the whole circuit and pins for touch but lacks the connections ... now i need to buy a different display :(

I even ordered and soldered the touch IC at the back myself ... completely senseless, yay ...
But at least i know what the problem is now, thanks a lot.

shirisha3020, 2020-03-22:
> I'm encouraged since you got it to work. My next step is to erase Arduino on this machine and reinstall as necessary to build the sketch. Maybe I'm picking up something from another library ... I [...]

Can you send me the code that you used to do a project with 2.8tft spi module and nodemcu?

shirisha3020, 2020-03-22:
Can you send me the code that you used to do a project with 2.8tft spi module and nodemcu?

eleleljo, 2020-11-12:
Hello
I also get no response of my 2.4" touch display by using ESP32 DEV Module
I use this WeatherAppArduiTouch https://github.com/Slaveche90/WeatherAppArduiTouch
I can load all the TFT_eSPI 320x240 examples. 
I get no response of the touch screen when I touch the Numbers.

In the User_Setup.h I wrote like the e-Book says:

#define TFT_MISO		19
#define TFT_MOSI		23
#define TFT_SCLK		18
#define TFT_CS			5   
#define TFT_DC			4   
#define TFT_RST			22  
//#define TFT_RST		-1  
//#define TFT_BL		32  
#define TOUCH_CS		14     
//#define TFT_WR		22  

this is the output of the Test_touch_controller

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1216
ho 0 tail 12 room 4
load:0x40078000,len:9720
ho 0 tail 12 room 4
load:0x40080400,len:6352
entry 0x400806b8
x: 0     y: 3286     z: 19 
x: 0     y: 3484     z: 21 
x: 0     y: 3181     z: 25 
x: 205     y: 3138     z: 5 
x: 0     y: 3499     z: 20 
x: 0     y: 3502     z: 23 
x: 0     y: 2392     z: 15 
x: 257     y: 2908     z: 25 
x: 0     y: 3341     z: 29 
x: 480     y: 3048     z: 4 
x: 0     y: 2714     z: 23 
x: 0     y: 3476     z: 21 
x: 80     y: 3183     z: 20 
x: 268     y: 2886     z: 19 
x: 112     y: 3595     z: 22 
x: 428     y: 3125     z: 29 
x: 41     y: 3263     z: 34 
x: 0     y: 3458     z: 5 
x: 0     y: 3483     z: 22 
x: 585     y: 2886     z: 27 
x: 50     y: 3190     z: 33 
x: 560     y: 2884     z: 23 
x: 368     y: 3086     z: 5 
x: 0     y: 2448     z: 14 
x: 0     y: 3448     z: 27 
x: 0     y: 3479     z: 27 
x: 0     y: 3469     z: 23 
x: 0     y: 2375     z: 16 
x: 0     y: 3424     z: 21 
x: 0     y: 3511     z: 20 

x: 394     y: 3152     z: 21 
x: 0     y: 3477     z: 22 
x: 0     y: 2402     z: 15 
x: 147     y: 3611     z: 24 
x: 0     y: 3351     z: 7 
x: 0     y: 3519     z: 24 
x: 0     y: 2361     z: 16 
x: 0     y: 3472     z: 24 
x: 0     y: 3436     z: 13 
x: 702     y: 2740     z: 19 
x: 0     y: 3456     z: 17 
x: 0     y: 2672     z: 19 
x: 0     y: 2368     z: 13 
x: 480     y: 3056     z: 4 
x: 4     y: 3488     z: 22 
x: 0     y: 3464     z: 24 
x: 80     y: 3260     z: 37 
x: 0     y: 2357     z: 11 
x: 0     y: 3450     z: 5 
x: 356     y: 3101     z: 5 
x: 77     y: 3247     z: 37 
x: 0     y: 3429     z: 21 
x: 0     y: 3376     z: 29 
x: 0     y: 3515     z: 21 
x: 0     y: 3515     z: 20 
x: 0     y: 3462     z: 24 
x: 0     y: 3351     z: 9 
x: 555     y: 2937     z: 27 
x: 0     y: 3448     z: 27 
x: 128     y: 3591     z: 24 
x: 0     y: 3480     z: 20 
x: 0     y: 3359     z: 18 
x: 1     y: 3484     z: 20 
x: 384     y: 3123     z: 5 
x: 0     y: 2408     z: 11 
x: 434     y: 3107     z: 37 
x: 0     y: 3480     z: 20 
x: 2     y: 3494     z: 32 
x: 0     y: 3520     z: 15 
x: 0     y: 3460     z: 69 
x: 662     y: 2597     z: 13 
x: 482     y: 3050     z: 4 

Dos someone has an idea what is wrong at my device?

Greetings
Josef

Bodmer, 2020-11-12:
The WeatherAppArduiTouch  sketch does not use the touch screen. The sketch name is misleading, it just uses the AdruiTouch case.

mattordev, 2024-05-28:
I know this is a dead thread, but currently running into a multitude of issue with touch, The library has been great in displaying things and has made life really easy, but touch input evades me! 

I'm using this [display](https://www.amazon.co.uk/dp/B0BTGNN9VV?psc=1&ref=ppx_yo2ov_dt_b_product_details)

It uses the IL9844 chip, I thought after realizing to disconnect the MISO and just connecting it to T_DO would do the trick, nope, now instead of spitting out random data, it gives me nothing. 

Also doing this on a RPi Pico W if it matters!

The diagnostic code gives me this:
x: 0     y: 0     z: 0 
x: 0     y: 0     z: 0 
x: 0     y: 0     z: 0 
x: 0     y: 0     z: 0 
x: 0     y: 0     z: 0 
x: 0     y: 0     z: 0 
x: 0     y: 0     z: 0 
x: 0     y: 0     z: 0 

At one point, I even gave up on SPI touch and tried IRQ, but couldn't figure that one out either. I feel like there's something I'm missing, but that I must be close!

Current wiring is like:
T_OUT > TFT_MISO (SDO)
T_DIN > TFT_MOSI (SDI)
T_CS > GPIO22
T_CLK > TFT_SCK
TFT_MISO (SDO) > T_OUT
TFT_LED > 3.3v
TFT_SCK > GPIO18
TFT_MOSI (SDI) > GPIO19
TFT_D/C > GPIO20
TFT_RST > GPIO21

Here's both my User_Setup_Select.h & Setup21_ILI9488.h:
[image]

If I recall correctly, I have completely changed the setup21 file, it displays great, just touch refuses to work, but if there's an actual setup file for the pico w please feel free to scream at me where it is.

Any pointers would be greatly appreciated, as man I am LOST.

Here's the current messy but correct (I think, triple checked) wiring: 

[image]
[image]
[image]

Wierdly enough I can get the x cord to move IF I connect up the MISO pin but I know that that's wrong:
x: 1097     y: 2368     z: 2273 
x: 1193     y: 2624     z: 2019 
x: 1033     y: 2576     z: 2587 
x: 2121     y: 2560     z: 2343 
x: 1097     y: 2112     z: 2307 
x: 1033     y: 2192     z: 2379 
x: 1033     y: 2112     z: 2555 

I think it's beaten me for tonight, it's currently slowly getting bright outside again. Let me know if there's any other info I can provide to get this fixed! If need be I can make a new issue too :)