Bodmer/TFT_eSPI #119: 320x240 cellular automata doesn't compile on m5stack

laurentopia, 2018-04-05:
`Error compiling for board M5Stack-Core-ESP32.`
I wouldn't assume it's a drop in but I saw m5stack in the ili9341 code so I assumed it'd work.

laurentopia, 2018-04-05:
When I uncomment 
#include <User_Setups/Setup12_M5Stack.h>           // Setup file for the ESP32 based M5Stack
and move the folder to inside libraries I now get these errors:
 error: redefinition of 'const unsigned char widtbl_f16 [96]' PROGMEM const unsigned char widtbl_f16[96] =         // character width table

does this mean I need to delete some de-doubled files ? which ones?

Bodmer, 2018-04-05:
You should not need to move or remove anything.

Load a fresh copy of the library and try again. Only one USer Setup should be included so make sure the selection looks like the attached.
[setup.zip](https://github.com/Bodmer/TFT_eSPI/files/1882191/setup.zip)

laurentopia, 2018-04-07:
finally got it - thanks

closed 2018-04-07 by Bodmer