Bodmer/TFT_eSPI #120: How to install for m5stack?

laurentopia, 2018-04-05:
I've tried moving it in the libraries folder and getting all sort of duplicate error.
Any tuto?

Bodmer, 2018-04-05:
I have both the M5Stack and TFT_eSPI libraries loaded and see no errors.

If you edit the library User_Setup_Select.h file so that only the M5Stack configuration is loaded then all should be fine. e.g.

#include <User_Setups/Setup12_M5Stack.h>

and comment out the line:

// #include <User_Setup.h>

The the User_Setup.h and User_Setup_Select.h header file comments tell you more if you want some customization.

 The M5Stack library itself is based partly on code from TFT_eSPI.

I do not see any errors when I compile for the M5Stack can you list some of the errors and tell me if the sketches compile even though warnings are listed? Are you using the Arduino IDE or PlatformIO?

laurentopia, 2018-04-07:
I hadn't commented out #include <User_Setup.h>, now that I have, it compiles, thank you!
Do you know if it batches multiple draw calls. What I mean is 100 calls to drawLine turn into 100 write to the cpu ram but only 1 transfer via SPI to the ILI? 
SPI -> ILI seems to be the bottleneck.

Bodmer, 2018-04-07:
No there is no batch handling sue to the way the screen protocol works. If you write the lines to a sprite then the lines go to RAM but then you may find writing the sprite to the TFT takes more time than writing 100 lines individually.

closed 2018-04-07 by Bodmer

laurentopia, 2018-04-07:
What about if you write 300 points or lines, doesn't the start/stop transaction signal x300+color data take more time than a full frame transfer?

Bodmer, 2018-04-07:
Iy you want to set just one pixel anywhere on the screen you need to send 11 control bytes to define the window to be written to and 2 bytes for the colour.  To fill rectangle or line you send 11 bytes then N x 2 bytes, where N is the number of pixels.  If you know the SPI clock rate you can do the maths quite easily.

So at 16MHz SPI rate in theory you can set a single pixel in 13*8/16 = 6.5 microseconds. For large blocks of pixels a the same SPI rate you can get near 1 microsecond per pixel (16 bits). In practice transfers are slower as the chip select and command/data line must be toggles too.