Bodmer/TFT_eSPI #121: same performances at 80M - why?

laurentopia, 2018-04-10:
This isn't an issue, but an oddity I noticed. When I uncomment SPI_FREQUENCY 80000000 the performances of the PDQ test are the same as 20000000, on an m5stack which I think is able to 40M at least, do you know why?

Bodmer, 2018-04-10:
This suggests the SPI rate is not being changed in the right file which is Setup12_M5Stack.h. The default for the M5Stack is 27MHz [here](https://github.com/Bodmer/TFT_eSPI/blob/master/User_Setups/Setup12_M5Stack.h#L200).

You can also change the SPI frequency in your sketch providing this does not conflict with any other SI perihperals. in the PDQ example you could add the following line after tft.init():

SPI.setFrequency(SPI_FREQUENCY);

Where you define the SPI_FREQUENCY in Hz, e.g. 20000000 for 20MHz.

The performance for a large rectangle fill tracks the SPI clock rate very well. this is a graph that shows the performance for a screen fill. As you can ee the Adafruit_GFX library performs poorly with increasing SPI rate because the processor is spending a lot of time preparing the next pixel so the performance scales poorly with SPI rate.

[https://i.imgur.com/nJK70ug.png](url)

Bodmer, 2018-04-10:
I have just done the test myself.

In the sketch SPI.setFrequency(SPI_FREQUENCY); only works on the ESP8266, this is linked to the transaction support being required on the ESP32 to switch the HAL mutex, so the library sets the frequency before the start of each SPI transaction.

For a single screen fill the result for an ESP32 at different SPI frequencies is:
20MHz -> 66904us
40MHz -> 36015us
80MHz -> 20439us

In comparison for the ESP8266 with the CPU locked at 160MHz:
20MHz -> 62527us
40MHz -> 31506us
80MHz -> 16293us

In conclusion the results change with SPI frequency on both processors. The ESP8266 has the performance edge when dedicated to screen updates, but the dual processors on the ESP32 will ultimately be better due to task sharing when WiFi needs to be supported as well.

closed 2018-04-10 by Bodmer

laurentopia, 2018-04-11:
Thanks for looking at this @Bodmer.
40M/s works now and I am getting similar speedup, but 80M/s white screens the m5 so maybe all m5 are not creates equal.
A nice speed up :)