Bodmer/TFT_eSPI #125: GfxUi (drawJpeg - can first argument be a String rather than const char?)

al-ang, 2018-04-15:
This is regarding the GfxUi code so I'm not sure if this issue fits here. 

Please make drawJpeg first arg a String to be consistent with drawBmp. 
    void drawBmp(String filename, uint8_t x, uint16_t y);
    void drawJpeg(const char *filename, int xpos, int ypos);  < current
    void drawJpeg(String filename, int xpos, int ypos);  < suggested

Background: While playing with the weather station demo I had to resize some of the icons, and in the process changed them from bmp to jpg.  Then, the code used to generate the filename broke when I used drawJpeg.  If the drawJpeg args were consistent with drawBmp, the demo logic would work as-is.

This might fly in the face of good C programming code (me == neophyte).  But, if possible, it would help those who use jpeg images.

Bodmer, 2018-04-21:
The simplest approach for me is to suggest you add a converting function. I think this will work but have not tried it:

void drawJpeg(String filename, int xpos, int ypos)
{
  int16_t len = filename.length() + 2;
  char buffer[len];
  filename.toCharArray(buffer, len);
  drawJpeg(buffer, xpos, ypos);
}

Pablo2048, 2018-04-21:
May I suggest another version?
void GfxUi::drawJpeg(String filename, int xpos, int ypos)
{
  drawJpeg(buffer.c_str(), xpos, ypos);
}

Bodmer, 2018-04-21:
@Pablo2048 
That's interesting, I did not think the Arduino environment supported c_str() but it does indeed. Thanks.

closed 2018-04-23 by Bodmer