Bodmer/TFT_eSPI #126: Can I examine current value of text dataum color, padding, and font?

glown, 2018-04-20:
Is there a way to determine the current TextDatum, TextColor, TextPadding, and Font?  

I have a function that is called on a timer, and changes made in the function are affecting calls made outside of the function.  So I think If I save current values, do my busuness, and then reset the values, I think it should stop random things from happening (like a value with the wrong font and size, etc).

al-ang, 2018-04-21:
Rather than do anything of substance in the function being called by the timer, set a flag in the timer.  Then, in the main loop check that flag and if set, then do what needs to be done.  This will ensure that everything is executed sequentially.  Not sure if it'll work for you, but thought it was worth mentioning.

glown, 2018-04-21:
GitHub sent an email with your suggestion so I saw it right away.  Since it was easy to implement I gave it a shot .... seems to be working so far.  I think this is a good solution for my problem.  Thank you!

closed 2018-04-21 by glown

Bodmer, 2018-04-21:
@glown 

This is a good question, for some reason I protected the library variables even though they can be set via functions.  I have now made those variables "public" and updated the Github repository, so you can now access them using the "dot" approach.

The bad news is that some of the variable names do not relate so well to the function name that sets them!

ISTR the Arduino style guidelines discourages the use of structures and "dot" access to variables to aid code readability for beginners, but using a struct would be a neat way of reading and restoring a group of settings.

The approach below is a temporary fix for you to try, I will work out a neater method at some point.

Examples to read the values: 

  uint16_t current_font      = tft.textfont;    // Current selected font
  uint16_t current_size      = tft.textsize;    // Current font size multiplier
  uint16_t current_datum     = tft.textdatum;   // Text reference datum
  uint16_t current_textcolor = tft.textcolor;   // Text foreground color
  uint16_t current_bgcolor   = tft.textbgcolor; // Text background color
  uint16_t current_padding   = tft.padX;        //Text padding in x direction

glown, 2018-04-21:
This is great.  Thank you!