Bodmer/TFT_eSPI #127: Can't get adafruit huzzah32 and featherwing tft (ILI9341 and HX8537D) to work

fritsjan, 2018-04-21:
I have everything working with the huzzah8266 and I am very happy with the results of loading images and the speed of this library.
I am loading jpgs from sd and all works fine.
Now I need to switch to the esp32 because of speed and more IO pins. I changed the configuration to the correct pins (which I tested with the standard adafruit libraries and confirmed working) but the screen stays blank.
The serialmonitor informs me images are loading and rendering (with even more promising rendering times) but no results on the screen. Not even the intro text to connect to the serial.
I tried to tweak settings like SPI speed, but no result.

Also all other examples like graphic tests do not work with this hardware combo (I have 2 sets of hardware so it is no fault in the hardware)

Looks to me it is something in the driver for the esp32

fritsjan, 2018-04-21:
Also, please add a clear example for loading both .bmp and .jpg from sd card. 
All examples are based on spiffs but I (and probably a lot of other people) need to easy switch graphics by overwriting them on the sd.

Bodmer, 2018-04-21:
Run [this sketch](https://github.com/Bodmer/TFT_eSPI/tree/master/examples/Test%20and%20diagnostics/Read_User_Setup) and post the results.

Which SD library are you using and which pins for the chip select.

Make sure the tft library instance is created before the SD library so the TFT chip select is initialised.

Bodmer, 2018-04-21:
Here is a SD card Jpeg example. It works fine on my screen with and ESP32.
[ESP32_SDcard_jpeg.zip](https://github.com/Bodmer/TFT_eSPI/files/1934841/ESP32_SDcard_jpeg.zip)

fritsjan, 2018-04-21:
Nope still same result.
But I see you say in your code must need gpio 5 (ss) for sd_cs.
Mine uses GPIO 14 (default for featherwing/esp32 combo)
And when I look at the gpio for the esp32 I see gpio 5 is actually SCK so this one is in use for the spi bus, right?

Here are my diagnostics:
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:812
load:0x40078000,len:0
load:0x40078000,len:11572
entry 0x40078a5c

[code]
Processor    = ESP32
Frequency    = 240 MHz
Transactions = Yes 
Interface    = SPI 
Display driver = 9341
Display width  = 240 
Display height = 320 

TFT_CS   = D15 (GPIO 15)
TFT_DC   = D33 (GPIO 33)
Font GLCD   loaded
Font 2      loaded
Font 4      loaded
Font 6      loaded
Font 7      loaded
Font 8      loaded
Smooth font enabled

Display SPI frequency = 40.0 MHz 
[/code]

fritsjan, 2018-04-21:
I don't think the problem lies with the SD lib. Because all other examples without SD also do not work. While the examples from the adafruit_ILI9341 lib (graphicstest_featherwing and fulltest_featherwing) do work out of the box (for both huzzah8266 and huzzah32)

Bodmer, 2018-04-21:
Map the DC pin to a gpio in the range 0-31. That should fix it. this is a restriction that I see is currently undocumented!

Bodmer, 2018-04-21:
There is a typo in the example, should be[ D3 to be compatible with the SD lib test example](https://github.com/espressif/arduino-esp32/blob/master/libraries/SD/examples/SD_Test/SD_Test.ino#L6).

Bodmer, 2018-04-21:
You can specify the SPI pins when SPIbeging() is called, defaults are:

SPI.begin(int8_t sck=-1, int8_t miso=-1, int8_t mosi=-1, int8_t ss=-1);

Bodmer, 2018-04-21:
Had a look at the library, you need to specify the pins in the user setup file since the Huzzah requires a different pin for SCLK:

#define TFT_SCLK 5
#define TFT_MISO 19
#define TFT_MOSI 23

fritsjan, 2018-04-22:
Mapping the DC pin on the tft board to pin D27 (next to the D33 pin) and setting this in the config did the job!
In the config I did not need to set the SPI pins, just left it uncommented. Also setting them to 19,18 and 5 also worked (as they are mapped now).
I am using D14 for the SD card pin. Works.

Thanks for your great support. Please document this somewhere for future uses.
Also add an clean example in the lib for reading bmp and jpg (and png?) from sd.
Took me some while to figure this out.
(like in the gfxui lib?)

fritsjan, 2018-04-22:
So this now works on esp32 feather  (huzzah32) and small featherwing tft (ILI9341 driver).
But no luck with the bigger 3,5" featherwing tft with the HX8357 driver.
Remapped DC to 27 also and #define HX8357D_DRIVER in setup

Bodmer, 2018-04-22:
Hi, great to know it works now.  I do not have one of those HX8357D screens but other users have reported it works but maybe that was with tge ESP8266. I woukd try using the reset pin in case garbage is being sent to the screen when the chip select is floating. Also run that diagnostic sketch again to verify the setup is as you expect.

With Adafruit products I usually suggest that folk stick to the Adafruit tutorials and libraries, as their step by step instructions makes life easy. I assume you have it configured for SPI and the SD Card interface is working?

Do post back when you have it working and if you find an issue with the TFT_eSPI or if any special setup is needed.

I can review how you have connected it up  if you post pictures.

wawibu, 2018-04-22:
I got finally my HX8357C running with my ESP8266. Used the following mapping:
IM0 / IM1 / IM2 all to +3.3V for 4-line SPI
SDI (MOSI) to D7 (GPIO13)
RESTET to RST
WR/SCL to D5 (GPIO14)
RS (DC) to D4 (GPIO2)
CS to D3 (GPIO0
LEDA to D8 (GPIO15) - to turn the backlight on / off, could also be fixed to +3.3V

was first testing with 3-wire SPI (IM0 / IM2 = Vcc + IM1 = GND) but that was not working. So I swtiched to 4-wire SPI and mapped the RS(DC) pin also and it worked fine.

Only change I need to make to switch my sketch from ILI9341 2.8" 320x280 to HX8357c 3.5" 480x320 was changing the selected driver within PlatformIO.

fritsjan, 2018-04-22:
I am using a HX8357D version (I checked through the adefruit driver) with standard '4' wire spi (miso, mosi, sck and cs and dc)

I just get a white screen, same like with the ILI9341 display. Mapped DC to 27 does not seem to help here.
Checked settings with the debug tool and all looks ok. (correct driver selected and correct pins).
Tried all examples for 480x320, none seem to work.
Tried with adafruit library and it works (also DC on pin 27)

Since I will be doing the same project for 2 different screens I guess I have to work with the ada lib now.
But that library is sooo slow with images... (loading a bmp takes 3500ms compared to 150ms for a jpg with your lib)

Unfortunately the reset pin for the tft is not available on the featherwing board, it is tied to the reset of the micro I think.

Bodmer, 2018-04-22:
@fritzjan
Can you post a web link to the display you have, there are a few variants of the HX8357 diver (B, C and D iterations, and they are not compatible.  Maybe the Adafruit driver reads the driver ID from the screen and chooses the right init sequence?

The HX8357D init sequence is taken from the Adafruit library and at least 2 users have it running with the display sold by Adafruit, but I have not tested it myself.  Are you saving TFT_eSPI works with the screen you have with the same driver settings but on an ESP8266 only?

fritsjan, 2018-04-22:
I am now using version 0.20.13, was using 0.20.11.
Now it works on esp32 and 3,5" screen... (and pin 33)
First set the spi speed to 20Mhz and graphicstest_one_lib works, later set it to 40Mhz and it still works... Weird...

Also other examples work now. jpg loading etc.
So probably the pin 33 issue fixed it. (or pin 27 caused issues also)

Thanks!!

Bodmer, 2018-04-22:
OK, I checked pin 27 is OK too so not sure what happened.

Anyway, great to hear you have it working. If any special changes to library were needed then let me know.

Also if you can post a copy of the setup file I can add it as a Huzzah feather setup option.

Good luck with your project.

closed 2018-04-22 by Bodmer

fritsjan, 2018-04-24:
[feather_huzzah_setup.zip](https://github.com/Bodmer/TFT_eSPI/files/1942418/feather_huzzah_setup.zip)
Here is my setup file.
Would be nice if you can implement something you can set in the code instead of changing a user_setup file.

Something like a #define HX8357 

Or even have the setup file local (near my sketch) and point to that one with an include or define.

Bodmer, 2018-04-24:
The Arduino IDE environment does not permit libraries to see #defines in the sketch, hence the rather awkward method used.

If you use PlatformIO you can keep the configuration in each project:

https://github.com/Bodmer/TFT_eSPI/tree/master/Tools/PlatformIO

fritsjan, 2018-04-25:
Ok #define won't work then, but you could give variables when initializing the driver. For example this is how it works with the neopixels:
Adafruit_NeoPixel pixels = Adafruit_NeoPixel(NUMPIXELS, PIN, NEO_RGB + NEO_KHZ800);

Where NEO_RGB declares which 'driver' to use (could also be NEO_RGBW for example)

Also when needed, but maybe not neccesary when doing above:
As you see in my user_setup file, I did a define there for #ifdef ESP32 and #ifdef ESP8266 those work and are processed by arduino when you select a board.
So this already takes care of half the work if you switch often between boards (not likely, but still)

But this can also be done in the sketch itself, for example:

#ifdef ESP32
   TFT_eSPI tft = TFT_eSPI(HX8357_32_ada);
   #define SD_CS 14
#endif
#ifdef ESP8266 
   TFT_eSPI tft = TFT_eSPI(HX8357_8266_ada);
   #define SD_CS 4
#endif

OR the selection could be done in the driver file

Bodmer, 2018-04-25:
The reason for the approach taken is that I have at least 6 different hardware combinations. So it is very simple for me to spend a few seconds changing the User_Setup_Select.h file (delete // add //) and then any of the examples will run with no changes at all. Given the number of examples and test sketches it would take much longer to edit each sketch I wanted to run.

So it is not ideal, but it works very well for me and beginners seem to like the lack of confusing configuration clutter in sketches.

Phando, 2018-11-26:
Hey All,
Great library and discussion thus far. I too am in the boat of trying to get the Huzzah32 to play nice with the Adafruit 3.5 display. Using the settings from @fritsjan I am seeing some success with the Adafruit 2.4" display, but the 3.5" display is only ever showing all white. The console logs still work and the app is running. Can you please let me know if you are seeing a successful setup for the 3.5" display with the following settings?

```
#define ILI9341_DRIVER
or
#define HX8357D_DRIVER

#define TFT_MISO 19
#define TFT_MOSI 18
#define TFT_SCLK 5
#define TFT_CS    15  // Chip select control pin
#define TFT_DC    33  // Data Command control pin
//#define TFT_RST   4  // Reset pin (could connect to RST pin)
#define TFT_RST  -1  // Set TFT_RST to -1 if display RESET is connected to ESP32 board RST
#define TOUCH_CS 32     // Chip select pin (T_CS) of touch screen - Not needed in my app

#define SPI_FREQUENCY  40000000 // Maximum to use SPIFFS
#define SPI_TOUCH_FREQUENCY  2500000
```

This library looks way better than what ships with the display and I would love to use it. 

Thanks!

Phando, 2018-11-26:
Hey All,
Great library and discussion thus far. I too am in the boat of trying to get the Huzzah32 to play nice with the Adafruit 3.5 display. Using the settings from @fritsjan I am seeing some success with the Adafruit 2.4" display, but the 3.5" display is only ever showing all white. The console logs still work and the app is running. Can you please let me know if you are seeing a successful setup for the 3.5" display with the following settings?

#define ILI9341_DRIVER
or
#define HX8357D_DRIVER

#define TFT_MISO 19
#define TFT_MOSI 18
#define TFT_SCLK 5
#define TFT_CS    15  // Chip select control pin
#define TFT_DC    33  // Data Command control pin
//#define TFT_RST   4  // Reset pin (could connect to RST pin)
#define TFT_RST  -1  // Set TFT_RST to -1 if display RESET is connected to ESP32 board RST
#define TOUCH_CS 32     // Chip select pin (T_CS) of touch screen - Not needed in my app

#define SPI_FREQUENCY  40000000 // Maximum to use SPIFFS
#define SPI_TOUCH_FREQUENCY  2500000
This library looks way better than what ships with the display and I would love to use it.

Thanks!

Bodmer, 2018-11-26:
The settings look OK.

Run the Test and diagnostics>Read_User_Setup example and see if the right settings are being picked up by the compiler. The sketch does not display anything on the screen but sends a report to the Serial Monitor window at 115200 baud.

Bodmer, 2018-11-26:
Make sure you have the latest 1.3.6 version (updated a few hours ago and now available in the Arduino IDE library manager).

fritsjan, 2018-11-26:
It has been a while, I worked with both the 2,5"and the 3,5" display as I recall (in different setups), which made it even more difficult. Find attached my setting files and my project file (which is for the 3,5" display)
I got this workin on TFT_eSPI version 0.20.13 which is a very old version by now ;-)

[eTFT_3_5.zip](https://github.com/Bodmer/TFT_eSPI/files/2617332/eTFT_3_5.zip)
[setup_files.zip](https://github.com/Bodmer/TFT_eSPI/files/2617318/setup_files.zip)

fritsjan, 2018-11-26:
> #define ILI9341_DRIVER
> or
> #define HX8357D_DRIVER

which one do you use? do you comment one out?

Phando, 2018-11-27:
I got things working with the HX8357D driver!
This is a very thorough library with tons of cool features, thanks!

Bodmer, 2018-11-27:
Glad to hear it is working. Thanks for the feedback.

Phando, 2019-06-04:
Revisiting this project, I am unable to show anything other than a white screen. I just picked up the latest and overwrote my config files (facepalm).

I am using the Adafruit Huzza32 and their 3.5" Featherwing Display (HX8357D).
 
TFT_eSPI.h is including only User_Setup_Select.h
User_Setup_Select is including User_Setups/Setup15_HX8357D.h

Setup15_HX8357D.h has the following definitions:
```
#define TFT_CS   	 15
#define TFT_DC  	 33
#define TOUCH_CS 32
#define SD_CS 	 14

My code thus far:
// Invoke TFT library
TFT_eSPI tft = TFT_eSPI();
TFT_eSprite spr = TFT_eSprite(&tft);

void setup()
{
  Serial.begin(115200); 

  tft.begin();
  tft.setRotation(2);  // 0 & 2 Portrait. 1 & 3 landscape
  tft.fillScreen(TFT_BLACK);
  spr.setColorDepth(16);
}
```

Any help with the config would be much appreciated.

GazTurtle, 2020-06-11:
Hi

GazTurtle, 2020-06-11:
HI Phando. I have My Adafruit Feather Huzzah 32 working with the 3.5 inch Feather Shield LCD 

https://learn.adafruit.com/adafruit-3-5-tft-featherwing

Here are my settings. The important discovery for me was setting a lower SPI frequency...

#define HX8357D_DRIVER
#define TFT_MISO 19
#define TFT_MOSI 18 // 23
#define TFT_SCLK 5 // 18
#define TFT_CS   15  // Chip select control pin
#define TFT_DC   33 // 2  // Data Command control pin
//#define TFT_RST   4  // Reset pin (could connect to RST pin)
#define TFT_RST  -1  // Set TFT_RST to -1 if display RESET is connected to ESP32 board RST

#define TOUCH_CS 32     // Chip select pin (T_CS) of touch screen
 #define SPI_FREQUENCY  20000000
#define SPI_READ_FREQUENCY  20000000
#define SPI_TOUCH_FREQUENCY  2500000

I haven't tested touch

Phando, 2020-06-11:
That's great news, I will give it a shot later this week.