Bodmer/TFT_eSPI #129: guru meditation when running graphicstest_one with a parallel ILI9488_DRIVER

laurentopia, 2018-04-26:
I finally got this gorgeous display, it's supposed to support the ILI9488 so I connected it, reqired the setup and compiled ok, but when it runs I get a guru meditation:

`Guru Meditation Error: Core  0 panic'ed (IllegalInstruction)
. Exception was unhandled.
Register dump:
PC      : 0x400d6a4b  PS      : 0x00060530  A0      : 0x800d1991  A1      : 0x3ffe3b30  
A2      : 0x3ffe3b58  A3      : 0x0000000a  A4      : 0x3ff49058  A5      : 0x00000000  
A6      : 0xf0014c00  A7      : 0xff000000  A8      : 0x00000002  A9      : 0x00000000  
A10     : 0x00000000  A11     : 0x00000000  A12     : 0x000000ff  A13     : 0x0000ff00  
A14     : 0x00ff0000  A15     : 0xff000000  SAR     : 0x0000000a  EXCCAUSE: 0x00000000  
EXCVADDR: 0x00000000  LBEG    : 0x400d2cc4  LEND    : 0x400d2d32  LCOUNT  : 0x00000000  

Backtrace: 0x400d6a4b:0x3ffe3b30 0x400d198e:0x3ffe3b50 0x400d2d59:0x3ffe3b90 0x400d1465:0x3ffe3bc0 0x400d8cff:0x3ffe3be0 0x400826ad:0x3ffe3c00 0x4008286d:0x3ffe3c30 0x400788a6:0x3ffe3c50 0x4007890d:0x3ffe3c80 0x40078a52:0x3ffe3cb0 0x40078b7a:0x3ffe3e70 0x40007c31:0x3ffe3eb0 0x4000073d:0x3ffe3f20

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x12 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:956
load:0x40078000,len:0
load:0x40078000,len:13076
entry 0x40078a58
user code done
`

`#define TFT_CS   9  // Chip select control pin
#define TFT_DC   7  // Data Command control pin - use a pin in the range 0-31
#define TFT_RST  8  // Reset pin

#define TFT_WR    26  // Write strobe control pin - use a pin in the range 0-31
#define TFT_RD    24

#define TFT_D0   14  // Must use pins in the range 0-31 for the data bus
#define TFT_D1   16  // so a single register write sets/clears all bits
#define TFT_D2   11
#define TFT_D3   10
#define TFT_D4   31
#define TFT_D5   30
#define TFT_D6   29
#define TFT_D7   28`

Bodmer, 2018-04-26:
Which ESP32 board are you using?

The problem appears to be that you are using pin numbers, not GPIO numbers. If you provide a link to your board I can identify the GPIO allocations to those pins.

laurentopia, 2018-04-26:
In the code comment you say "must use pin#0-31"... did you mean GPIO then?

It's this one:
https://www.amazon.com/gp/product/B079PVCF2G/ref=oh_aui_detailpage_o03_s00?ie=UTF8&psc=1

laurentopia, 2018-04-28:
I see, I corrected that and noticed that it wouldn't upload so I moved it away from the TX/RX pins which I think is used for usb or maybe eprom. Now it's uploading, the screen is lit up but there is nothing onscreen, just a faint rhythmical pulsing.
Note: this LCD has no DC but it has RS

final mapping:
`// ESP32 pins used
#define TFT_RST  36  // Reset pin
#define TFT_CS   39  // Chip select control pin
#define TFT_DC   25  // RS? Data Command control pin - use a pin in the range 0-31
#define TFT_WR   26  // Write strobe control pin - use a pin in the range 0-31
#define TFT_RD   32

// Must use pins in the range 0-31 for the data bus so a single register write sets/clears all bits
#define TFT_D0   4  
#define TFT_D1   15
#define TFT_D2   23
#define TFT_D3   22
#define TFT_D4   21
#define TFT_D5   19
#define TFT_D6   18
#define TFT_D7   5`

Bodmer, 2018-04-28:
The rhythmical pulsing implies that the TFT is being written to but the wrong driver is being used and thus the TFT is not being initialised,

The link to the TFT lists lots of potential driver chips, which are supplied randomly. We need to try and identify what driver is fitted. If I can find time I will write a diagnostic sketch that will read the TFT ID. That may provide the clue needed to select the right driver. One option is to try each driver with the screen and find out experimentally which driver is fitted, but it could be that the screen is fitted with a driver that would need new init code to be added to the library.

laurentopia, 2018-04-29:
I just tried the 2 other controllers with the same pin assignment from your lib and get the same pulsing.
Which wires initialize the LCD?
Could the lack of DC in this screen be a problem? "Register Select" doesn't really sound like "Data Command"

Bodmer, 2018-04-29:
The DC and RS signals are just different names for the same thing. They select whether a Data or Command register is to be accessed.

laurentopia, 2018-04-29:
ok so connections are a-ok, I tried all controllers. Any other idea? Maybe this LCD only takes 5V signal?

laurentopia, 2018-04-29:
I tried, it doesn't compile. I'll wait for your diag tool, it'll be interesting to see what the source of it was. Bet is on 5V vs 3.3V

Bodmer, 2018-04-30:
Here is the register read sketch.  It reads 256 registers, some may contain a TFT ID that is recognisable.

The display and ESP32 run from 3,3V so that is not the problem.

If you have wired the display using wires to the ESP32 then you may not get it working due to poor grounding (due to wire inductance) and glitches on the signals.  It is best to use an UNO style ESP32 board with these shields to maintain signal integrity.

Try the sketch and post the results.
[ESP32_Read_Registers.zip](https://github.com/Bodmer/TFT_eSPI/files/1962414/ESP32_Read_Registers.zip)

laurentopia, 2018-04-30:
Thanks.
Here is the result:
```
Register 0x00: 0x00000000
Register 0x01: 0x00000000
Register 0x02: 0x00000000
Register 0x03: 0x00000000
Register 0x04: 0x00000000
Register 0x05: 0x00000000
Register 0x06: 0x00000000
Register 0x07: 0x00000000
Register 0x08: 0x00000000
Register 0x09: 0x00000000
Register 0x0A: 0x00000000
Register 0x0B: 0x00000000
Register 0x0C: 0x00000000
Register 0x0D: 0x00000000
Register 0x0E: 0x00000000
Register 0x0F: 0x00000000
Register 0x10: 0x00000000
Register 0x11: 0x00000000
Register 0x12: 0x00000000
Register 0x13: 0x00000000
Register 0x14: 0x00000000
Register 0x15: 0x00000000
Register 0x16: 0x00000000
Register 0x17: 0x00000000
Register 0x18: 0x00000000
Register 0x19: 0x00000000
Register 0x1A: 0x00000000
Register 0x1B: 0x00000000
Register 0x1C: 0x00000000
Register 0x1D: 0x00000000
Register 0x1E: 0x00000000
Register 0x1F: 0x00000000
Register 0x20: 0x00000000
Register 0x21: 0x00000000
Register 0x22: 0x00000000
Register 0x23: 0x00000000
Register 0x24: 0x00000000
Register 0x25: 0x00000000
Register 0x26: 0x00000000
Register 0x27: 0x00000000
Register 0x28: 0x00000000
Register 0x29: 0x00000000
Register 0x2A: 0x00000000
Register 0x2B: 0x00000000
Register 0x2C: 0x00000000
Register 0x2D: 0x00000000
Register 0x2E: 0x00000000
Register 0x2F: 0x00000000
Register 0x30: 0x00000000
Register 0x31: 0x00000000
Register 0x32: 0x00000000
Register 0x33: 0x00000000
Register 0x34: 0x00000000
Register 0x35: 0x00000000
Register 0x36: 0x00000000
Register 0x37: 0x00000000
Register 0x38: 0x00000000
Register 0x39: 0x00000000
Register 0x3A: 0x00000000
Register 0x3B: 0x00000000
Register 0x3C: 0x00000000
Register 0x3D: 0x00000000
Register 0x3E: 0x00000000
Register 0x3F: 0x00000000
Register 0x40: 0x00000000
Register 0x41: 0x00000000
Register 0x42: 0x00000000
[... 166 lines cut ...]
Register 0xE9: 0x00000000
Register 0xEA: 0x00000000
Register 0xEB: 0x00000000
Register 0xEC: 0x00000000
Register 0xED: 0x00000000
Register 0xEE: 0x00000000
Register 0xEF: 0x00000000
Register 0xF0: 0x00000000
Register 0xF1: 0x00000000
Register 0xF2: 0x00000000
Register 0xF3: 0x00000000
Register 0xF4: 0x00000000
Register 0xF5: 0x00000000
Register 0xF6: 0x00000000
Register 0xF7: 0x00000000
Register 0xF8: 0x00000000
Register 0xF9: 0x00000000
Register 0xFA: 0x00000000
Register 0xFB: 0x00000000
Register 0xFC: 0x00000000
Register 0xFD: 0x00000000
Register 0xFE: 0x00000000
Register 0xFF: 0x00000000

```

Bodmer, 2018-05-01:
Not good!! Something is wrong with the connections, power supply or configuration.

Also if you have used connection wires you may have glitch problems on signals. I suggest you get an UNO format ESP32 and use a standard setup. Bear in mind that some ESP32 pins are input only and you will not get a compiler warning.

laurentopia, 2018-05-01:
Alright, we have a return!
What is that register thing testing?

Bodmer, 2018-05-01:
It reads registers in the TFT, typically some of these include an ID that helps identify the controller.

closed 2018-05-01 by Bodmer

laurentopia, 2018-05-02:
I connected to another display and this one's backlight turns on, here is the result:

```
rst:0x1 (POWERON_RESET),boot:0x16 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:956
load:0x40078000,len:0
load:0x40078000,len:13076
entry 0x40078a58
Registers from 0 to FF
Register 0x00: 0xFFFFFFFF
Register 0x01: 0xFFFFFFFF
Register 0x02: 0xFFFFFFFF
Register 0x03: 0xFFFFFFFF
Register 0x04: 0xFFFFFFFF
Register 0x05: 0xFFFFFFFF
Register 0x06: 0xFFFFFFFF
Register 0x07: 0xFFFFFFFF
Register 0x08: 0xFFFFFFFF
Register 0x09: 0xFFFFFFFF
Register 0x0A: 0xFFFFFFFF
Register 0x0B: 0xFFFFFFFF
Register 0x0C: 0xFFFFFFFF
Register 0x0D: 0xFFFFFFFF
Register 0x0E: 0xFFFFFFFF
Register 0x0F: 0xFFFFFFFF
Register 0x10: 0xFFFFFFFF
Register 0x11: 0xFFFFFFFF
Register 0x12: 0xFFFFFFFF
Register 0x13: 0xFFFFFFFF
Register 0x14: 0xFFFFFFFF
Register 0x15: 0xFFFFFFFF
Register 0x16: 0xFFFFFFFF
Register 0x17: 0xFFFFFFFF
Register 0x18: 0xFFFFFFFF
Register 0x19: 0xFFFFFFFF
Register 0x1A: 0xFFFFFFFF
Register 0x1B: 0xFFFFFFFF
Register 0x1C: 0xFFFFFFFF
Register 0x1D: 0xFFFFFFFF
Register 0x1E: 0xFFFFFFFF
Register 0x1F: 0xFFFFFFFF
Register 0x20: 0xFFFFFFFF
Register 0x21: 0xFFFFFFFF
Register 0x22: 0xFFFFFFFF
Register 0x23: 0xFFFFFFFF
Register 0x24: 0xFFFFFFFF
Register 0x25: 0xFFFFFFFF
Register 0x26: 0xFFFFFFFF
Register 0x27: 0xFFFFFFFF
Register 0x28: 0xFFFFFFFF
Register 0x29: 0xFFFFFFFF
Register 0x2A: 0xFFFFFFFF
Register 0x2B: 0xFFFFFFFF
Register 0x2C: 0xFFFFFFFF
Register 0x2D: 0xFFFFFFFF
Register 0x2E: 0xFFFFFFFF
Register 0x2F: 0xFFFFFFFF
Register 0x30: 0xFFFFFFFF
Register 0x31: 0xFFFFFFFF
Register 0x32: 0xFFFFFFFF
Register 0x33: 0xFFFFFFFF
Register 0x34: 0xFFFFFFFF
Register 0x35: 0xFFFFFFFF
Register 0x36: 0xFFFFFFFF
Register 0x37: 0xFFFFFFFF
Register 0x38: 0xFFFFFFFF
[... 175 lines cut ...]
Register 0xE8: 0xFFFFFFFF
Register 0xE9: 0xFFFFFFFF
Register 0xEA: 0xFFFFFFFF
Register 0xEB: 0xFFFFFFFF
Register 0xEC: 0xFFFFFFFF
Register 0xED: 0xFFFFFFFF
Register 0xEE: 0xFFFFFFFF
Register 0xEF: 0xFFFFFFFF
Register 0xF0: 0xFFFFFFFF
Register 0xF1: 0xFFFFFFFF
Register 0xF2: 0xFFFFFFFF
Register 0xF3: 0xFFFFFFFF
Register 0xF4: 0xFFFFFFFF
Register 0xF5: 0xFFFFFFFF
Register 0xF6: 0xFFFFFFFF
Register 0xF7: 0xFFFFFFFF
Register 0xF8: 0xFFFFFFFF
Register 0xF9: 0xFFFFFFFF
Register 0xFA: 0xFFFFFFFF
Register 0xFB: 0xFFFFFFFF
Register 0xFC: 0xFFFFFFFF
Register 0xFD: 0xFFFFFFFF
Register 0xFE: 0xFFFFFFFF
Register 0xFF: 0xFFFFFFFF
```

Bodmer, 2018-05-02:
Still not good!
GPIO34 to 39 are input only so you need ro reallocate those pins and run the test again.
See pin diagram here:
https://learn.sparkfun.com/tutorials/esp32-thing-hookup-guide

Note GPIO marked with *

laurentopia, 2018-05-02:
Thank you for your help, so many diagrams are incomplete.
This is the board:
https://raw.githubusercontent.com/playelek/pinout-doit-32devkitv1/master/pinoutDOIT32devkitv1.png
I moved away from 39... 35 and 34, it's like an obstacle course combined with knitting

```
// ESP32 pins used
#define TFT_RST  14  // Reset pin
#define TFT_CS   12  // Chip select control pin
#define TFT_DC   25  // Data Command control pin - use a pin in the range 0-31
#define TFT_WR   26  // Write strobe control pin - use a pin in the range 0-31
#define TFT_RD   32

// Must use pins in the range 0-31 for the data bus so a single register write sets/clears all bits
#define TFT_D0   4  
#define TFT_D1   15
#define TFT_D2   23
#define TFT_D3   22
#define TFT_D4   21
#define TFT_D5   19
#define TFT_D6   18
#define TFT_D7   5
```

Bodmer, 2018-05-02:
Run the Read_User_Setup and look at the Serial output to check the compiler is picking up the configuration you think it is.
Can you post pictures of your wiring?

laurentopia, 2018-05-02:
I ran it and got something unexpected: the setup file points to the 9488 and this says 9341, the pins are that of 9488 and 9341 has the wrong pins, the specs of the screen is 480x320 and this says 320x240.
I changed pins in 9341 setup file and pointed to that one, screen still blank.
```
[code]
Processor    = ESP32
Frequency    = 240 MHz
Transactions = Yes 
Interface    = Parallel 
Display driver = 9341
Display width  = 240 
Display height = 320 

TFT_CS   = D12 (GPIO 12)
TFT_DC   = D25 (GPIO 25)
TFT_RST  = D14 (GPIO 14)

TFT_WR   = D26 (GPIO 26)
TFT_RD   = D32 (GPIO 32)

TFT_D0   = D4 (GPIO 4)
TFT_D1   = D15 (GPIO 15)
TFT_D2   = D23 (GPIO 23)
TFT_D3   = D22 (GPIO 22)
TFT_D4   = D21 (GPIO 21)
TFT_D5   = D19 (GPIO 19)
TFT_D6   = D18 (GPIO 18)
TFT_D7   = D5 (GPIO 5)

Font GLCD   loaded
Font 2      loaded
Font 4      loaded
Font 6      loaded
Font 7      loaded
Font 8      loaded
Smooth font enabled

[/code]
```
[image]
[image]
[image]

Bodmer, 2018-05-09:
The connections look good, but the wires are far too long and will likely cause problems due to the number of signals that are toggling and the lack of a low impedance 0V return.  I will try to wire up a display like this but I am doubtful it will work, the TFT_WR line is a critical signal so any ringing on the signal edges due to wire inductance and reflections will cause problems as the logic edges have a high dV/dt on the ESP32. Maybe some series resistance will be required to damp any ringing so I will see if this is needed.

Bodmer, 2018-05-09:
In the photo, it looks like D12 is not connected to TFT_CS... check your wiring and compare to the GPIO allocations in the library.

laurentopia, 2018-05-13:
The pins were offset by one, I corrected this.
the register read ino now reads
`Registers from 0 to FF
Register 0x00: 0xFFFFFFFF
`
all the way to FF