Bodmer/TFT_eSPI #131: Third sprite not showing up (esp32)

laurentopia, 2018-04-29:
I declare 3 sprites: 320x40x16, 320x100x16 and 320x100x16
Only the first two show up
If I resize all of them to 256x then all three show up
So I thought too memory used, but all three combined are only 153k byte, the esp32 has 4mb of sram and ... the app runs
Any idea?

Bodmer, 2018-04-29:
The RAM on the ESP32 is divided up amongst different processes and thus only about 96K is made available for arrays and Sprites. The first two Sprites need about 90K so the third Sprite cannot be allocated space and thus is not created. The createSprite function returns a NULL pointer to indicate this has happened.

closed 2018-04-29 by Bodmer

laurentopia, 2018-04-30:
Which part of your code limits this allocation?

Bodmer, 2018-04-30:
You will have to Google the answer as it i not a restriction impossed by the TFT_eSPI library and it is not something I have investigated. Probably it is down to a division of Stack RAM space to different functions and is imposed by the underlying ESP32 IDF, probably configured in the SDK so that is the place to start looking.