Bodmer/TFT_eSPI #132: dark side of the moon

wysiecki, 2018-04-30:
Hi all,

got a short question about the moon in weather example.
When it runs for a longer period of time, the moon is getting distorted and finally disappears.

Any hint on this ?

THX
Martin

Bodmer, 2018-04-30:
Interesting! Have you allocated 3M to SPIFFS?

wysiecki, 2018-05-03:
Hi, yes, and also tried to convert the images to JPEG to save space.
And the images are correct, no faults, but still, check please the image.

Any idea ?

Thank you
Martin
[image]

wysiecki, 2018-05-03:
OK ! got my new nodeMCU today, and all is working fine, must have been some kind of bad chip.

THX

closed 2018-05-03 by wysiecki