Bodmer/TFT_eSPI #136: Uno esp32 and shield wiring

laurentopia, 2018-05-05:
"to the ESP32 analogue pins 35, 34 and 36 which are input only. To solve this I linked in the 3 spare pins IO15, IO33 and IO32 by adding wires to the bottom of the board as follows:"
What happens to 35 34 36? They gets shorted?

Bodmer, 2018-05-06:
34, 35 and 36 become unused inputs. They are not shorted or damaged as it is OK to drive then with normal logic levels, normally this might be another device driving them. As the pins can only be inputs there can be no drive state conflicts.

closed 2018-05-06 by Bodmer

laurentopia, 2018-05-06:
Oh I see, so I get the data of 15,33,32 if I analogRead them.

Bodmer, 2018-05-06:
Yes, if 15, 33 or 32 are set as output. If 15, 33 or 32 are set as input the you could use the 34, 35, 36 pins to read analogue values.