Bodmer/TFT_eSPI #138: Multiple displays and documentation

codebude, 2018-05-07:
At first let me say thank you for this great library. I've got two questions/problems:

1. Is it possible to use multiple displays on one microcontroller? (I'm using an ESP-32 and technically a second display could share the SPI bus with the first display and also there are enough free pins to use as/for CS. But since ports are defined in User_Setup.h and not in the libs constructor I don't see, how to create a second instance of the class for another display/with other ports.)
2. Maybe I'm just blind, but is there any kind of documentation/wiki for this lib? I'm thinking about a page/document that summarizes all methods of the lib and briefly describes their function and parameters.

Bodmer, 2018-05-07:
Thanks for the feedback. To answer your questions:
1. Yes, this is a constraint. The library evolved from one that I adapted fir AVR processors where economy of RAM and FLASH use was the driver for the approach used as I wanted control of the features. These contraints are eliminated with the resiurces available in the ESP8266 and ESP32! I have run two displays on AVR processors by not defining TFT_CS in the setup and selecting the TFT to write to in the sketch by controlling two chip sekect lines from the sketch. Not very convenient compared to two instances but it worked well.
2. Yes, there is no Wiki or manual at the moment, it is on my TO DO list but unfortunately I spend too much time adding new features and with paid work! I will see if I can move it up the priority list, thanks for prompting me to do this!

codebude, 2018-05-07:
The solution for (1) sounds like a good option for now. I'll adapt it for my use-case. Regarding the documentation - maybe this will help near-term: http://arduinodocs.readthedocs.io/en/latest/ (I'm not affiliated with this project, nor have I used it. But at first sight it looked promising.)

Bodmer, 2018-05-08:
In the setup file comment out the line that defined TFT_CS then the library will not control any pin for the chip select.  If the two chip select lines are say pin 4 and pin 5, then set both high and set just one low to write to a particular display.  In the setup you can make both CS lines low for the tft.init();

If you want to clear both screens at the same time set both CS lines low and do a tft.fillScreen();

To write to just one display then just set one CS line low.

Note that if you set the cursor position or text color etc for one display then you will need to set it again for the second display.

closed 2018-05-08 by Bodmer

bayerla, 2020-06-23:
Dear Bodmer, I am not so family with github, and programming. But it was nice to start my TTGO-T-Display projekt with many thanks to your TFT_eSPI.cpp Ich understand your time gap for documentation.
Attached I made a excel analyse of TFT_eSPI.cpp for myself. ( without realy understand the Code ) 
[TFT_eSPI.cpp. All.pdf](https://github.com/Bodmer/TFT_eSPI/files/4822114/TFT_eSPI.cpp.All.pdf)

Someone can only learn by reading the overview of the Funktions and the Uses Variables and find fast the line in TFT_eSPI.cpp 
for seeing all. Feel free to add it anywere you think it my bee usefull, or proposals to change something or delete here.

ArnieO, 2020-09-08:
@bayerla 
Thank you for sharing your PDF; very helpful!!

Bodmer, 2020-09-08:
I have posted the function summary [here](https://github.com/Bodmer/User_Manual_TFT_eSPI), thanks.

Sampozzo, 2021-03-26:
Hello, sorry if i reopen this issue, i found only this place talking about two display. I was able to do alternate CS pins. But what about two different drivers? Is it possible to use ST7735 and ILI9431 togheter? If yes, how? 

Thank you very much!