Bodmer/TFT_eSPI #139: appropriate wiring for an ILI9341 touch screen RESET

aliceand, 2018-05-08:
I've seen people connect  RESET of a touch ILI9341 in two ways -- either share RESET with VCC and LED, or connect it to D4.  

Is one way correct and the other wrong?  And if using D4, what functionality does it provide?  I mostly want to do the smart thing, but I can see times when I'm running out of I/O pins and might want to use D4 for something else..

Bodmer, 2018-05-08:
You have a number of options:

1. Connect the display RESET line high (logic 1) and rely on the software command reset that the library uses during the init().
2. Connect the RESET line to the Arduino RST line so it gets reset when the Arduino board does
3. Connect the RESET line to a digital pin and let the library toggle it during init()

The 3rd option uses up a pin and as the RESET line is only used by the library during the init() function then it is a waste of a pin.

aliceand, 2018-05-08:
Thank you for a complete explanation.  I actually understand.

closed 2018-05-08 by aliceand