Bodmer/TFT_eSPI #140: graphicstest_one doesn't fill the 480x320 mcu friend

laurentopia, 2018-05-15:
I caved in and got a r32 board, soldered the 3 wires in the back and now it works!
But there is static below, white on the right and the display size is smaller than 480x320, I use the 480x example.
What's the solution?
[image]
[image]

laurentopia, 2018-05-15:
I switched to the smaller 320x240 screen and it works perfectly.
I wonder how eSPI gathers the information on screen size because it's nowhere in the .ino

Bodmer, 2018-05-15:
The screen driver and resolution are defined in [the setup](https://github.com/Bodmer/TFT_eSPI/blob/master/User_Setups/Setup2_ST7735.h#L27) and/or [driver code.](https://github.com/Bodmer/TFT_eSPI/blob/master/TFT_Drivers/ILI9481_Defines.h)

closed 2018-05-15 by Bodmer