Bodmer/TFT_eSPI #143: ST7735 INITB is mirrored

brownjd, 2018-05-19:
I modified ST7735_Rotation.h to handle my screen. It was sold at a ST7735R, but I'm guessing it is a B because no other combination worked. I noticed INITB is not explicitly handled, so I added condition for INITB. The issue I was trying to handle is that no matter what rotation I set, the text was reversed, as in a mirror.

Being new to Git I'm not sure how I'm supposed to propose changes, so I'm opening an issue. I'm happy to push, branch, or pull, or whatever if you're good with it. Just let me know. 

Here are my edits.

ST7735_Rotation.h

case 0:
 } else if(tabcolor == INITB) {
       writedata(TFT_MAD_MX | TFT_MAD_BGR);
     }

case 1:
} else if(tabcolor == INITB) {
       writedata(TFT_MAD_MV | TFT_MAD_MX | TFT_MAD_MY | TFT_MAD_BGR);
     }

case 2:
} else if(tabcolor == INITB) {
       writedata(TFT_MAD_MY | TFT_MAD_RGB);
     }

case 3:
} else if(tabcolor == INITB) {
       writedata(TFT_MAD_MV | TFT_MAD_RGB);
     }

Bodmer, 2018-05-19:
Thanks, I have updated the files to include your suggested code.

closed 2018-05-19 by Bodmer

brownjd, 2018-05-20:
I think TFT_RGB should be used in all 4 cases.

Bodmer, 2018-05-20:
OK, thanks I blindly copied your code and did not notice the differences. Thanks for pointing this out.