Bodmer/TFT_eSPI #144: Support multiple configurations

brownjd, 2018-05-19:
I would like to compile a single ESP image that handles multiple screens. I want to decide at run time which screen is connected (via my own configuration file) and then init appropriately. With Adafruit's original ST7735 lib this is easy because the tab type is passed in to initR. Using defines in this library makes it problematic.

Any way to handle this?

Bodmer, 2018-05-25:
I see the problem, this is not something I had envisaged would be needed. I will park this one and see if I can look at adding this but it will be a while due to paid work.

Bodmer, 2018-05-25:
It was simple to add, so I have updated the library.

Use line in form:
tft.init(1);
or:
tft.init(INITR_REDTAB);

// Enumerated configurations already defined in the library are:
#define INITR_GREENTAB 0x0
#define INITR_REDTAB 0x1
#define INITR_BLACKTAB 0x2
#define INITR_GREENTAB2 0x3
#define INITR_GREENTAB3 0x4
#define INITR_GREENTAB128 0x5
#define INITB 0xB

al-ang, 2018-05-31:
This change makes my projects much more manageable.  Thank you.