Bodmer/TFT_eSPI #146: readPixel not working on esp8266 +HX8357D 480x320 display

perlix, 2018-05-25:
I'm trying to switch from Adafruit's HX8357D library to TFT_eSPI in a sketch that runs on a Wemos D1 R2 and drives an HX8357D 480x320 display.

Displaying bmp files from SD with drawBmp() is much faster now (!) and drawPixel() works fine. Only readPixel() doesn't seem to work. Calling it turns the entire display white. The sketch continues normally and there are no esp8266 errors in Serial monitor.

Any clues or tips for debugging? 
Thanks - Paul

Bodmer, 2018-05-25:
I don't have one of those displays so am unable to test. I cannot see anything wrong with the code. Does the Adafruit's HX8357D library support reading the display?

Are you using the SPI interface?

perlix, 2018-05-25:
Thanks for the quick reply.

I'm using the SPI interface of the display and hardware SPI.
The Adafruit library itself has no readPixel, but they published an external readPixA() function:

```
uint16_t readPixA(int x, int y) {
  tft.setAddrWindow(x,y,x,y);
  digitalWrite(TFT_DC, LOW);
  digitalWrite(TFT_CLK, LOW);
  digitalWrite(TFT_CS, LOW);
  tft.spiwrite(0x2E); // memory read command
  digitalWrite(TFT_DC, HIGH);
  uint16_t r = 0;
  r = tft.spiread(); // discard dummy read
  r = tft.spiread() >> 3; // red: use 5 highest bits (discard three LSB)
  r = (r << 6) | tft.spiread() >> 2; // green: use 6 highest bits (discard two LSB)
  r = (r << 5) | tft.spiread() >> 3; // blue: use 5 highest bits (discard three LSB)
  digitalWrite(TFT_CS, HIGH);
  return r;
}
```
spiread() and spiwite() are part of the Afafruit library:
```
uint8_t Adafruit_HX8357::spiread(void) {
  uint8_t r = 0;
  if (_sclk == -1) {
    SPI.beginTransaction(spi_settings);
    r = SPI.transfer(0x00);
    SPI.endTransaction();              // release the SPI bus
  } else {
    for (uint8_t i=0; i<8; i++) {
      digitalWrite(_sclk, LOW);
      digitalWrite(_sclk, HIGH);
      r <<= 1;
      if (digitalRead(_miso))
	r |= 0x1;
    }
  }
  return r;
}

void Adafruit_HX8357::spiwrite(uint8_t c) {
  //Serial.print("0x"); Serial.print(c, HEX); Serial.print(", ");
  if (_sclk == -1) {
    // hardware SPI!
    SPI.transfer(c);
  } else {
    *clkport &= ~clkpinmask;
    //digitalWrite(_sclk, LOW);
    for(uint8_t bit = 0x80; bit; bit >>= 1) {
      if(c & bit) {
	//digitalWrite(_mosi, HIGH); 
	*mosiport |=  mosipinmask;
      } else {
	//digitalWrite(_mosi, LOW); 
	*mosiport &= ~mosipinmask;
      }
      //digitalWrite(_sclk, HIGH);
      *clkport |=  clkpinmask;
      //digitalWrite(_sclk, LOW);
      *clkport &= ~clkpinmask;
    }
  }
}
```
I extracted spiread() and spiwrite() from the library, but when I call readPixA(), the esp8266 crashes.

A solution would be very welcome, because yours is the first library that brings speed to this great display.

Paul

Bodmer, 2018-05-25:
The code posted has some missing variable definitions... it performs the same function as [the code in the TFT_eSPI library](https://github.com/Bodmer/TFT_eSPI/blob/master/TFT_eSPI.cpp#L584-L600). Without one of those displays I am unable to help.

perlix, 2018-05-25:
Thanks anyway. I'll keep trying because it's a great library.