Bodmer/TFT_eSPI #147: drawing pixel with transparency

HumamHelfawi, 2018-05-25:
Hi,

Thanks for the great library!

Is it possible to draw pixel over an already drawn background with transparency?
Such as:
`tft.drawPixel(5,5,tft.color565(255,200,10), 65); // 65% transparency`

Thanks!

Bodmer, 2018-05-25:
You will need a display that works with the pixelRead() function to read the display pixel, then use the alphaBlend() function to combine that pixel colour with the new colour and then repaint the pixel with the blended colour using drawPixel().

HumamHelfawi, 2018-05-25:
I see. Thank you very much for the quick reply!

Bodmer, 2018-05-25:
You can also read and blend pixels if a Sprite but then you are restricted on the size of the area you can work with.

closed 2018-05-25 by Bodmer