Bodmer/TFT_eSPI #148: Arduino compatible ILI9488 library for ESP32

ZYENA, 2018-05-28:
Hello,
We are trying to interface ILI9488 based 3.5" TFT LCD with ESP32 board. We have two LCDs.
1.  [https://item.taobao.com/item.htm?spm=a230r.1.14.92.20ac5b37CRDNRy&id=532194921387&ns=1&abbucket=4#detail](url)
2.  [https://item.taobao.com/item.htm?spm=a230r.1.14.8.299f4ed7JGMBgP&id=556843872127&ns=1&abbucket=12#detail](url)

Library used for making this LCD working with **ESP32 is Bodmer/TFT_eSPI** and IDE use for same is Arduino IDE. By using this library, LCD worked fine in parallel mode but in SPI mode Both LCDs shows dotted display and not worked.
here attaching library used and photo of TFT output in spi mode. 
[image]
[image]

[TFT_eSPI-master.zip](https://github.com/Bodmer/TFT_eSPI/files/2044758/TFT_eSPI-master.zip)

We want to interface ILI9488 with ESP32 dev board. Please give us solution for this. 

Thanks & Regards,
Zyena

Bodmer, 2018-05-28:
Not sure is this will help.

Make sure the display is in 4 wire SPI mode (uses DC pin) and reduce the SPI clock rate to 20MHz

ZYENA, 2018-05-29:
okey

ZYENA, 2018-05-29:
Can we change color depth from 16 bit to 18 bit? Because in 4 wire SPI mode, one pixel data contains 18-bit color depth information.

Bodmer, 2018-05-30:
I've had a look at the ILI9488 data sheet and this display requires 18 bit colors for the 4 wire SPI mode, these bits must be split across 3 serial bytes. At the moment the TFT_eSPI library does not support this color mode.

closed 2018-05-30 by Bodmer

fredericplante, 2018-05-30:
I guess you could transform the 565 data format into 666 data format, would not be much job to do.