Bodmer/TFT_eSPI #149: drawLine not working if x < y

wawibu, 2018-05-28:
try to draw a graph line and need therefore draw multiple small lines. As soon as the width of the line is smaller than the height, the function draws nothing.

eg. 
drawLine(5,156,15,156,TFT_White); << works fine
drawLine(5,156,15,170,TFT_White); << draws nothing

tried this also with wider ranges, but the issue is the same. So if x width would be 20 and y height would be 21 it will not draw a line.

Bodmer, 2018-05-28:
ESP32 or ESP8366?
SPI or parallel?
Which TFT driver?

wawibu, 2018-05-29:
sorry - missed that
ESP8266
4 wire SPI (40Mhz - tried 20Mhz as well without any differences)
HX8357D
tried on a second ESP and second display but is the same. 
[image]

Bodmer, 2018-05-29:
Interesting, it appears that the drawLine is failing on the HX8357D display even though it works fine on other displays I have.  It looks like the lines that would be classed as "steep" are not being drawn which is strange. I will think on this further as it seems peculiar to the display you have, unfortunately I do not have one of that type to try. Are you sure this is not a bug in your sketch? Can you provide a simple sketch that demonstrates the problem.

Bodmer, 2018-05-29:
OK, I have made a change to the TFT_eSPI.cpp file, try downloading that file fro GitHub and replacing the one in your library, try the test sketch again. Let me know if this does or does not fix it.

Bodmer, 2018-05-29:
Thanks for reporting this and helping to solve this issue.  Hope your project proceeds without further problems!

Hamed-Elahi, 2021-06-02:
Hello,

can you tell me what the problem was?

I have the same problem with this LCD "HY32D". The driver is for ILI9325.

When the code reads the ID of the LCD, then it prints ILI9326.