Bodmer/TFT_eSPI #152: ILI9488  with FT6236 touch screen problem

momosh13, 2018-06-04:
Hi
I have this screen ILI9488 with FT6236 touch screen and I'm trying to use TFT_eSPi to drive the screen with ESP32, the screen is working fine in 8bit parallel mode, but I'm having problem operating the touch part with FT6236 chip, from the data sheet I have to connect the i2c lines SDA / SCL and   interrupt signal INT, and when I'm running the i2c_scanner program, it shows I have a device connected on address 38, that means the touch screen is connected properly. When checking the i2c line the is no data is coming out but there is interrupt signal when I'm touching the screen.
The screen is from buydisplay.com and this is the data shits for the screen
http://www.buydisplay.com/download/ic/FT6236-FT6336-FT6436L-FT6436_Datasheet.pdf
http://www.buydisplay.com/download/manual/ER-TFTM035-6_Datasheet.pdf
Please help me to use this touch screen with this library.
Thank you

Bodmer, 2018-06-06:
The TFT_eSPI library only supports the SPI based XPT2046 touch controller. To use a FT6236 you will need to find a library that supports that controller such as [this one by Adafruit](https://github.com/adafruit/Adafruit_FT6206_Library). Adafruit also have tutorials that may help.

The touch screen library will report touch positions that can then be used to reference graphics coordinates for drawing on the screen etc.

I do not have a touch screen with that controller so I am unable to help.

closed 2018-06-10 by Bodmer

j45p41, 2020-12-23:
> The TFT_eSPI library only supports the SPI based XPT2046 touch controller. To use a FT6236 you will need to find a library that supports that controller such as [this one by [...]

Hi @Bodmer, I have been using your library in my project with LGVL but the resistive display that I was using was not fit for purpose. I have since ordered a number of screens and just realised that they use FT6236. So before I panic, can you tell me if you mean that you are recommending using the Adafruit library ONLY for the touchscreen or to control the touch and the display? 

I can see from [](https://github.com/DustinWatts/FT6236) that it is possible to use the touch independently.  If so then how could this be implemented with TFT_eSPI. Thanks in advance!

Bodmer, 2020-12-23:
Yes, I am suggesting that only the Adafruit touch library is used with TFT_eSPI. You may find other better libraries but this is not something I have investigated. Once you have it running then it would be great if you reported a "how to" in the [discussions tab](https://github.com/Bodmer/TFT_eSPI/discussions).

j45p41, 2020-12-23:
> Yes, I am suggesting that only the Adafruit touch library is used with TFT_eSPI. You may find other better libraries but this is not something I have investigated. Once you have it running then it [...]

Many thanks and I will.

marcelstoer, 2023-02-28:
I also have a ILI9488 & FT6236 display.

> > it would be great if you reported a "how to" in the [discussions tab](https://github.com/Bodmer/TFT_eSPI/discussions).
>
> Many thanks and I will.

@j45p41 looks like this never happened. Did you not succeed or succeed but not write about? Looks like the only discussion about the FT6236 was started by someone else at https://github.com/Bodmer/TFT_eSPI/discussions/1700.