Bodmer/TFT_eSPI #153: Smooth Fonts

JLuisAlcaraz, 2018-06-07:
Hi,
Thank you for your contribution to this excellent library.
I'm interested in using smooth fonts, but I dont know how to use it. Pleas, can you put an example of the process to convert a font to vlw.

Thanks in advance!

Bodmer, 2018-06-08:
Load the [Processing IDE from here](https://processing.org/download/).

Run [the sketch from here in](https://github.com/Bodmer/TFT_eSPI/tree/master/Tools/Create_Smooth_Font) the Processing IDE.  Edit the sketch to choose different fonts and set the sizes of the created vlw file.

You can also use the "Create Font" option which is in the Processing IDE Tools menu but this is less flexible than the sketch for determining the characters to be included.

Bodmer, 2018-06-09:
Let me know if you need more help.

closed 2018-06-09 by Bodmer