Bodmer/TFT_eSPI #154: ESP32_SDcard_jpeg.ino Pinout

anlpereira, 2018-06-08:
HI Bodmer, can you share the pinout diagram of this example code? I'm using the generic ESP32 board and ILI9341 LCD. Thanks

Bodmer, 2018-06-09:
Assuming you have the display working then the SD card SPI signals are connected to the same SPI lines as the display, except for the chip select (CS) pin. The SD card CS pin has to be defined in your sketch.

If you are running the example sketch "ESP32_SDcard_jpeg.ino" where ESP32 pin 15 is the TFT chip select, then the wiring is:
SD_CS > ESP32 pin 5
SD_MOSI > ESP32 pin 23
SD_MISO > ESP32 pin 19
SD_SCK > ESP32 pin 18

anlpereira, 2018-06-09:
Att.,I have the LCD running using the Adafruit_GFX library, not yours. I
didn't make it work with your library. So, there are some functions that
are not implemented.

Anderson Lima Pereira
Engenheiro Eletrônico
Reg. Nac. 270398134-1
+55 (79) 99917-2053