Bodmer/TFT_eSPI #155: ili9488 SPI support

Mike-the-one, 2018-06-10:
hi, @Bodmer we are working on a change to make this lib compatible with ili9488 in SPI mode, we would like to submit a pull request, are you interested? :-)

momosh13, 2018-06-13:
Good idea!!!
We can save lot of pin's, and also ili9488 on 16 bit mode has better colors then on 8 bit mode.
Thank you

Bodmer, 2018-06-13:
I have made a few changes to the library to make it compatible with the ILI9488 SPI based interface. I am waiting for a display board to arrive to test the updates before release. Due to paid work pressures it will be a few weeks before the next release.

Mike-the-one, 2018-06-16:
@Bodmer Thank you so much!  We made the draw line / string / bitmap working but break bitmap on other lCDs... so think the best is to stop now, and use your official code ... since you already have the code done, do you mind send your modified code to us, so we could give it a try first? :-)

Mike-the-one, 2018-06-19:
also when is scheduled next release?

Bodmer, 2018-07-07:
Support for 4 wire SPI based ILI9488 displays has been added.  This has been tested on an ESP8266 and ESP32. The defaul SPI rate in the setup is 27MHz but my display runs OK at 40MHz.

The ILI9488 in SPI interface mode supports 18 bit colours which are sent as 3 bytes, so the screen updates will be slower than a 16 bit colour SPI display.

momosh13, 2018-07-08:
The new SPI/ILI9488 is working perfect, little slower then 8bit parallel, but it's a new world.
Thank you !!!!

Bodmer, 2018-07-08:
Thanks for the feedback!

closed 2018-07-08 by Bodmer

Mike-the-one, 2018-07-19:
@Bodmer just wondering does the touch work?  We have ili9488 with XT2640 based touch, the display works, but not the touch.  Can you please confirm?  thanks