Bodmer/TFT_eSPI #156: Regression?

MarcFinns, 2018-06-11:
Hi Bodmer, way back we had a long discussion and you eventually allowed calling init() to soft reset the display WITHOUT resetting the SPI interface. This was useful in pin-constrained projects that use MISO for something else. An example is ([ATMOSCAN](https://www.instructables.com/id/AtmoScan/).
This was working fine in version 0.18.10. 
Today I updated the library to the latest version and nothing works anymore. Or better, it works fine at the beginning but if the init() method is called later, the display goes blank. As I am using GPIO12 to control the backlight, I suspect you have a regression with init() resetting the SPI interface and reassigning the pins to itself.
Any chances you could have a look? 
THANK YOU

Bodmer, 2018-06-13:
Not sure what happened there I will look into this and add it back in. Not sure when it got stripped out but I can look back though the history.

Bodmer, 2018-07-06:
I have added this back in.

closed 2018-07-06 by Bodmer

MarcFinns, 2018-07-06:
Thanks, will test it soon!

MarcFinns, 2018-07-07:
It works! Thanks again