Bodmer/TFT_eSPI #159: Display Stops Working when T_CLK is connected in Overlap Mode

generalannoyance, 2018-06-16:
Hey -- basically title. Screen works fine in non-overlap, but in overlap as soon as I connect the T_CLK it stops working. I can leave everything else connected and it'll draw fine. I've double checked  wiring and connectivity. 
Here's my config: 
[image]

Thanks in advance for looking at it.

Bodmer, 2018-06-18:
A restriction of overlap mode is that only one SPI device can be connected, so a touch screen controller cannot also be connected.  This restriction is because in overlap mode the FLASH memory SPI bus is being used so the processor must have hardware control is all SPI devices via the chip select, and there is only one available hardware chip select.

closed 2018-06-18 by Bodmer

generalannoyance, 2018-06-18:
Ahh okay, fair enough. Is this mentioned in the documentation anywhere or did I just miss it? Thanks for the response though.