Bodmer/TFT_eSPI #160: Problem with Waveshare 4inch Touch LCD Shield for Arduino (Not RPi model)

emilesprenger, 2018-06-16:
I can't seem to get this display to work.
https://www.waveshare.com/product/modules/oleds-lcds/arduino-lcd/4inch-tft-touch-shield.htm
I'm using a Wemos D1 Light
It uses a ILI9486 controller, and I've tried RPI_ILI9486_DRIVER, but it's not working.
I've also tried RPI_ILI9486_DRIVER, but that also doesn't work.
When using RPI_ILI9486_DRIVER, I've lowered the SPI speed.

Nothing seems to work ..

Any thoughts?

Thanks,

Emile

Bodmer, 2018-07-06:
I have looked at the display and the schematic. The schematic is similar to the RPi model but appears to use the same clocking approach.  The SPI speed to use is 20MHz.

I will test the library whith my RPi display again to make sure the library still works with that display.

emilesprenger, 2018-07-07:
Thanks .. lemme know how it goes .. i'll keep trying over here to find out what's wrong

Bodmer, 2018-07-25:
The RPi screen I have works fine with the latest library so that is not the issue.  I do not have the same screen which you have so unfortunately I am unable to help further,

closed 2018-07-25 by Bodmer