Bodmer/TFT_eSPI #163: support the IPS 1.3" ST7789 with CS pin to ground?

Nulista, 2018-06-18:
Hello Bodmer,

See display in link below (Aliexpress) This display CS pin is always connected to ground.
But is not working with the library. How can i fix it?
thx 
Ludo

https://nl.aliexpress.com/item/High-Definition-1-3-Inch-IPS-TFT-Display-IPS-LCD-Screen-1-3-inch-3-3V/32861867084.html?spm=a2g0z.search0104.3.15.7df27bf72ZAASz&ws_ab_test=searchweb0_0,searchweb201602_3_10152_10151_10065_10344_10068_5723115_5722815_10342_10343_10340_5722915_10341_5722615_10696_10084_10083_10618_10304_10307_10820_10821_10302_5722715_10843_10059_100031_10103_10624_10623_10622_5722515_10621_10620,searchweb201603_50,ppcSwitch_5&algo_expid=a77fb30f-beb9-4c7f-a7da-d30685f63fd7-2&algo_pvid=a77fb30f-beb9-4c7f-a7da-d30685f63fd7&priceBeautifyAB=0

Bodmer, 2018-07-06:
I have added support for that board.

It is a bit odd, but SPI mode 3 must be used when the chip select is not used. Other displays use SPI mode 0

Nulista, 2018-07-07:
Hello,

 

Thx a lot,

 

It works perfect!!

 

Closed #163 <https://github.com/Bodmer/TFT_eSPI/issues/163> .

RafaelScience, 2018-08-31:
Hello friend could share your library with me? I have the same problem

Bodmer, 2018-08-31:
The library supports the ST7789 with chip select held low. Use the Setup18_ST7789.h

RafaelScience, 2018-09-01:
I m using Esp 32, 

 Setup18_ST7789.h

#define TFT_MISO 19
#define TFT_MOSI 23
#define TFT_SCLK 18
#define TFT_CS   LOW  // Chip select control pin
#define TFT_DC    2  // Data Command control pin
#define TFT_RST   4

Bodmer, 2018-09-01:
Comment out the chip select define thus:

//#define TFT_CS

garyinbg, 2019-02-23:
I'm just going to go and try this, well on Monday anyway :)
Thanks in advance Bodmer. :)

Can I just ask why SPI mode 3 please

Bodmer, 2019-02-25:
Some displays with the ST7789 controller have CS wired low and these only work with mode 3 for some reason.

garyinbg, 2019-02-26:
Ok, many thanks for the reply :)

tablatronix, 2019-07-26:
Does this mean these displays cannot share spi with other devices?

Bodmer, 2019-07-26:
Yes, one way around this is to add additional circuitry to, for example, block the SPI clock line (SCK) to the TFT when another device is selected.

tablatronix, 2019-07-26:
Thats a good idea, I was just messing around trying to find some solution.

Bodmer, 2019-07-27:
A gotcha... The ST7789 is unusual, it is operated in SPI [Mode 3](https://github.com/Bodmer/TFT_eSPI/blob/master/TFT_eSPI.h#L41), so the SCK line must be driven high when you do not wish to communicate with the TFT,  otherwise an extra clock pulse will be introduced and that will mess up the next command.

tablatronix, 2019-07-27:
I had some other issues with this module, I could not use esp8266 reset, I had to use a IO for rst, and one that did not output on startup, might have something to do with the spi mode3 also

garyinbg, 2019-07-27:
I'm sorry for delay, thank you, yes... Appreciate the information 😊Gary RowlandsTel:   +359 884 252869Mail: rowlandsgary@yahoo.com -------- Original message --------From: Shawn A <notifications@github.com>Date: Sat, Jul 27, 2019, 8:19 PMTo: Bodmer/TFT_eSPI <TFT_eSPI@noreply.github.com>Cc: Gary Rowlands <rowlandsgary@yahoo.com>, Comment <comment@noreply.github.com>Subject: Re: [Bodmer/TFT_eSPI] support the IPS 1.3" ST7789 with CS pin to ground? (#163)I had some other issues with this module, I could not use esp8266 reset, I had to use a IO for rst, and one that did not output on startup, might have something to do with the spi mode3 also

Bodmer, 2019-07-29:
I think the problem could be:
1. When using the power on reset the TFT pins are floating thereafter until the code boots and sets them into a defined state. During the hiatus period the TFT sees garbage commands and clock pulses and appears to "lock up". Subsequently it does not accept commands, probably because there has not been a multiple of 8 clock pulses so the commands are all bit shifted.  Normally the CS line will re-sync the shift register to the 8 clock pulse sequence.

or:

2. The reset pulse is too short so it does not meet the display data sheet timing values causing the display to not initialize correctly.

I always use a pin to drive the reset line on displays that have CS tied low.

tablatronix, 2019-07-29:
2 is my suspect, I have been testing on my logic analyzer just to better understand it, I have seen some people implement an autoreset circuitry, I am trying to find more info about that.

I also lifted the CS trace and added my own cs tap, it makes the display much more useful

garyinbg, 2019-07-29:
Ah that's a very interesting addition, can I ask how you went about that (cs tap) and what advantages have you found? GGary RowlandsTel:   +359 884 252869Mail: rowlandsgary@yahoo.com -------- Original message --------From: Shawn A <notifications@github.com>Date: Mon, Jul 29, 2019, 5:05 AMTo: Bodmer/TFT_eSPI <TFT_eSPI@noreply.github.com>Cc: Gary Rowlands <rowlandsgary@yahoo.com>, Comment <comment@noreply.github.com>Subject: Re: [Bodmer/TFT_eSPI] support the IPS 1.3" ST7789 with CS pin to ground? (#163)2 is my suspect, I have been testing on my logic analyzer just to better understand it, I have seen some people implement an autoreset circuitry, I am trying to find more info about that.
I also lifted the CS trace and added my own cs tap, it makes the display much more useful

Bodmer, 2019-07-29:
I connected up an ST7789 display to a NodeMCU board and used the RST line as the reset source with the DC line from the display to pin D3 (GPIO 0) and when powered up the TFT failed to initialise. Pressing the reset button also failed to initialise the display indicating the issue is not the length of the reset pulse. This leaves spurious SPI clock pulses as a potential culprit.

To stabilise the SCK line before the software has booted connect I added a 10K pullup from VCC (3V3) to the SCK line and the display still failed to initialise. I changed to a 10K pull down on the SCK line and the display ran fine when pressing the reset button, it also booted and ran fine when the NodeMCU is powered up with a DC supply.

This setup did fail at startup when plugged into a PC for the first time as soon as the PC enumerates the COM port to the NodeMCU.  Moving the TFT DC line to the NodeMCU pin D1 fixed this.  This is possibly due to the dual functionality of D3 (GPIO 0) for the ESP8266 during boot.

So in summary to use the RST line from the NodeMCU I found:

1. A pull down (e.g. 10K) must be fitted to the TFT SCK line.
2. Do not use D3 (GPIO 0) for the DC line

With this setup the display worked reliably using the hardware RST line.  If this also works for you then I am interested to know if this is a general fix for these tied down CS displays.

garyinbg, 2019-07-29:
Many thanks for both emails 😊Gary RowlandsTel:   +359 884 252869Mail: rowlandsgary@yahoo.com -------- Original message --------From: Bodmer <notifications@github.com>Date: Tue, Jul 30, 2019, 12:35 AMTo: Bodmer/TFT_eSPI <TFT_eSPI@noreply.github.com>Cc: Gary Rowlands <rowlandsgary@yahoo.com>, Comment <comment@noreply.github.com>Subject: Re: [Bodmer/TFT_eSPI] support the IPS 1.3" ST7789 with CS pin to ground? (#163)Reopened #163.

tablatronix, 2019-07-29:
I will test, I did have to move dc around on another display I will check a few , ones that do not pulse on boot especially

tablatronix, 2019-08-04:
I can confirm the sck pulldown works, I am not having an issue with D3 DC but it might be unique to vcc or something, probably wise to avoid it still, I will keep testing it for a bit.

closed 2019-08-10 by Bodmer