Bodmer/TFT_eSPI #165: Text fits not in screen/does not clear screen

biejay, 2018-06-24:
Hey Guys,

i just hooked up my **ESP8266 to an ST7735 128x128** TFT Screen ([ST7735 from ebay](https://www.ebay.de/itm/1-44-Colorful-SPI-TFT-LCD-Display-ST7735-128X128-Replace-Nokia-5110-3310-LCD/382121587331?ssPageName=STRK%3AMEBIDX%3AIT&_trksid=p2060353.m2749.l2649))

I did connect everything as mentioned in the User_setups\Setup2_st7735.h file. 
(LED=Vin
SCK=D5
SDA=D7
A0=D3
RST=D4
CS=D8
GND=GND
VCC=3V3)

But when i'm trying your digital clock file for example, it **doesnt fit on screen properly**? also i got the problem, that the old texts **wont get completely cleared** out. 
Could you please help me with that?

[image]

biejay, 2018-06-25:
UPDATE:
I'm super dumb, had to **change the selected driver in the user_setup.h file to ST7735** obviously.
The **grafic error** in the upper right seems to come **from the font**. changed font and now everything looks great!

Bodmer, 2018-06-25:
The example is written for a different size screen e.g 128x160, so the digits will not fit the 128x128 screen you have.

I will create some examples for these smaller screens at some point.

closed 2018-06-25 by Bodmer