Bodmer/TFT_eSPI #166: drawing text on top of jpeg image shows text bg color

aliceand, 2018-06-26:
I draw a jpeg image on the screen, and then write some text over the image.  Problem is that the background configured with setTextColor shows.  I'm not sure how to describe this, but is it possible to draw text on top of a jpeg image and see only the text?  I think I'm asking to be able to set a clear background in setTextColor?

This is for a splash screen, so it's not that critical.  If I cannot write the text, I might integrate the text into the background jpeg and then just drawJpeg.  However, there are times I want to have dynamic text to show status with a graphical background even during the splash (connecting, loading, calculating, ...).

Bodmer, 2018-06-26:
Use setTextColor with one parameter, this parameter being the text foreground colour you wish to use.

aliceand, 2018-06-26:
Cool!  Thank you.

closed 2018-06-26 by aliceand