Bodmer/TFT_eSPI #167: Support for TFT based on ILI9486, 8-bit parallel

migueltk, 2018-06-27:
I want to support this TFT, https://es.aliexpress.com/item/3-2-inch-TFT-LCD-screen-module-Ultra-HD-320X480-for-Arduino-MEGA-2560-R3-Board/32605255967.html?spm=a219c.10010108.1000016.1.474b1359PHPlgq&isOrigTitle=true
 using an ESP32 module LoLin32 v1.0.0, someone in Aliexpress evaluations indicates that it has achieved it, but in parallel mode with the TFT_eSPI library. I have tried it, I created a new "user setup", Setup19_ILI9486_Parallel.h with the following content:

```
#define ESP32_PARALLEL
#define RPI_ILI9486_DRIVER

// ESP32 pins used
#define TFT_CS   33  // Chip select control pin
#define TFT_DC   15  // Data Command control pin - use a pin in the range 0-31
#define TFT_RST  32  // Reset pin

#define TFT_WR    4  // Write strobe control pin - use a pin in the range 0-31
#define TFT_RD    2

#define TFT_D0   12  // Must use pins in the range 0-31 for the data bus
#define TFT_D1   13  // so a single register write sets/clears all bits
#define TFT_D2   26
#define TFT_D3   25
#define TFT_D4   17
#define TFT_D5   16
#define TFT_D6   27
#define TFT_D7   14

#define LOAD_GLCD   // Font 1. Original Adafruit 8 pixel font needs ~1820 bytes in FLASH
#define LOAD_FONT2  // Font 2. Small 16 pixel high font, needs ~3534 bytes in FLASH, 96 characters
#define LOAD_FONT4  // Font 4. Medium 26 pixel high font, needs ~5848 bytes in FLASH, 96 characters
#define LOAD_FONT6  // Font 6. Large 48 pixel font, needs ~2666 bytes in FLASH, only characters 1234567890:-.apm
#define LOAD_FONT7  // Font 7. 7 segment 48 pixel font, needs ~2438 bytes in FLASH, only characters 1234567890:.
#define LOAD_FONT8  // Font 8. Large 75 pixel font needs ~3256 bytes in FLASH, only characters 1234567890:-.
//#define LOAD_FONT8N // Font 8. Alternative to Font 8 above, slightly narrower, so 3 digits fit a 160 pixel TFT
#define LOAD_GFXFF  // FreeFonts. Include access to the 48 Adafruit_GFX free fonts FF1 to FF48 and custom fonts

// Comment out the #define below to stop the SPIFFS filing system and smooth font code being loaded
// this will save ~20kbytes of FLASH
#define SMOOTH_FONT
```

It is based on the contents of the file "Setup16_ILI9488_Parallel.h".

In the file "User_Setup_Select.h" I commented on the line "#include <User_Setup.h>" and included a new line "#include <User_Setups / Setup19_ILI9486_Parallel.h>".

I have connected the TFT device but it does not work.

Some help.

Bodmer, 2018-06-28:
I am going to add the ILI9486 SPI mode support but do not have a display to test it with. I will be updating the repository within the next 10 days with SPI support for the ILI9488 as well.

migueltk, 2018-06-28:
Thanks for answering.

Great, this is incredible! I have tried the drivers and definitely the best one is the 'ILI9488_DRIVER', I have tried them with the example 'UTF_Demo_480x320' in both the time 'runtime msecs' is similar but with the driver 'ILI9481_DRIVER' in the central red box can be seen the lines that compose it, while with the driver 'ILI9488_DRIVER' the red color is uniform.

Greetings.

wilhelmzeuschner, 2018-07-07:
@migueltk this _someone_ in the feedback-section is actually me!

A brief explanation of how I got it to work:
`#define ILI9481_DRIVER` as previously suggested by @Bodmer 
And then you just have to define the pins you are going to use:
```
        #define ESP32_PARALLEL

	// The ESP32 and TFT the pins used for testing are:
	//Pins not connected to ESP32:
	#define TFT_RD    18  //prev. 35  Read strobe control pin  - use a pin in the range 0-31 -> pull up resistor, not connected
	#define TFT_RST  19  //prev. 32 Reset pin, toggles on startup		 -> pull up resistor, not connected
	
	//Used Pins:
	#define TFT_DC   15  // (RS) Data Command control pin - use a pin in the range 0-31	
	#define TFT_WR    32  // Write strobe control pin - use a pin in the range 0-31
	#define TFT_CS   33  //Chip select control pin (library pulls permanently low) -> pull down resistor?
	
	#define TFT_D0   27  // Must use pins in the range 0-31 for the data bus (prev. 12)
	#define TFT_D1   12  // so a single register write sets/clears all bits.
	#define TFT_D2   13  // Pins can be randomly assigned, this does not affect
	#define TFT_D3   14  // TFT screen update performance.
	#define TFT_D4   16
	#define TFT_D5   17
	#define TFT_D6   25
	#define TFT_D7   26
```
Please note that these are not the defaults. I have added some comments so that I can understand what I did in a few weeks or months into the future.

Bodmer, 2018-07-07:
The library has been updated, should now support parallel ILI9486 displays but I am unable to test.

closed 2018-07-07 by Bodmer